EdgeTX/edgetx #1488: tpro - functions switches playing havoc with autosource

pfeerick, 2022-01-23:
When testing #1479 I encounted this, but also repo'ed this in a nightly build for the TPro - but not in the OTX build so something probably went wrong in the initial import of TPro support.

Basically, autosource is locking on with active function switch - i.e. go to inputs, mixes, source, move something - everything works cool. Now press one of the function switches (assuming left at default 2POS mode), it will get picked as the source (ok), but stays active in as far as it keeps jumping back to the FS when you move any other switches or controls.

raphaelcoeffic, 2022-10-08:
@pfeerick is this issue still up-to-date? it seems to me it has been fixed in a PR which did not close the issue, right?

pfeerick, 2022-10-09:
I haven't checked it recently, but I'm pretty sure it is still present... just haven't had enough motivation to try and whack it...

PR #2505 (2022-10-12): fix(tpro): function switches auto-switch & position matching
Fixes #1488 

Summary of changes:
- fix auto-switch for function switches
- fix matching function switch UP position
commit e7969e73 (2022-10-12): fix(tpro): function switches auto-switch & position matching (#2505)

* fix(tpro): auto-switch for function switches

Fixes #1488

* fix(tpro): matching function switch UP position
commit 364c88cc (2022-10-12): fix(tpro): function switches auto-switch & position matching (#2505)

* fix(tpro): auto-switch for function switches

Fixes #1488

* fix(tpro): matching function switch UP position
closed 2022-10-12 by pfeerick