EdgeTX/edgetx #1489: V2.6 - model images are no longer the right resolution

piwi3910, 2022-01-23:
After upgrading to v2.6 rc2 i noticed that none of my model images where no longer filling the whole widget.
i used to make my images following these resolutions:

https://oscarliang.com/custom-splash-background-model-images/
so in my case my model images where: 135 × 114

it looks like now in v2.6 the model images need a higher resolution, then 114px could you tell us, what the new resolution is, as i couldn't see any changes in the release logs that were hinting a change was needed.

pfeerick, 2022-01-24:
What makes you think they need to be a different resolution in 2.6? I use images that are exactly 192×114 or nearly exactly that if the aspect ratio is wrong... 

v2.5.0
[image]

v2.6.0 (after turning on background and setting to white as that is an option now, as is model name size)
[image]

[image]

piwi3910, 2022-01-24:
I do the same
And now with v2.6 they are smaller on the screen, and as you can see the text above it also bigger

[EXTERNAL EMAIL]

[image][user-images.githubusercontent.com]<https://urldefense.com/v3/__https:/user-images.githubusercontent.com/5500713/150780948-c77b9e0f-8d4b-4263-b8ac-3dcabc021c09.png__;!!LpKI!1j-2TWfQ88r-pV7_iFprqsG2h6VMOdZgsIyPSPM2ZTK_tKDFXkGlZWi0Ho8uZXPnlg$>

[image][user-images.githubusercontent.com]<https://urldefense.com/v3/__https:/user-images.githubusercontent.com/5500713/150781170-ebe32bbf-0ea9-47c5-b6e5-76013eca15d2.png__;!!LpKI!1j-2TWfQ88r-pV7_iFprqsG2h6VMOdZgsIyPSPM2ZTK_tKDFXkGlZWi0Ho88npjQjg$>

[image][user-images.githubusercontent.com]<https://urldefense.com/v3/__https:/user-images.githubusercontent.com/5500713/150781328-43aa3cc0-7b4f-47fe-a2b9-2b30dd20d62a.png__;!!LpKI!1j-2TWfQ88r-pV7_iFprqsG2h6VMOdZgsIyPSPM2ZTK_tKDFXkGlZWi0Ho8X5KAkBQ$>

-

Internal Use - Confidential

pfeerick, 2022-01-24:
Right, so there are a couple things happening. With the 2.6 model widget, I removed the icon, and the line, but left the image size the same. The text size and color is configurable. The background color can also be changed, or turned off, if you want transparency. 

i.e. this is the same model image, but now different text size and color... 
[image]

You'll need to be more specific as I am not seeing any change in size of the actual model image - perhaps only the background fill making the widget seem smaller if you don't have the background fill turned on.

closed 2023-02-26 by gagarinlg