EdgeTX/edgetx #1491: Frsky D16 Protocol s.port Telemetry Missing

richardclli, 2022-01-24:
I have a friend used the RM receiver R161, and used the S.Port to send back telemetry, the included figures are from OpenTX 2.3.9 and compared to EdgeTX 2.6.0 released version, EdgeTX has a lot of sensors missing.  And he tried EdgeTX 2.5.0 as well.  Similar result, a lot of sensors missing.

[image]

[image]

I am thinking if the Frsky Hub telemetry is broken from day 1 in the OpenTX 2.4 branch?

And this is not related to https://github.com/EdgeTX/edgetx/issues/1139

**General information**
 - EdgeTX 2.6.0 release
 - Hardware RM TX16S

pfeerick, 2022-01-24:
But possible correlation / link to https://github.com/EdgeTX/edgetx/issues/1260

CNzhangsir, 2022-01-24:
This problem is the problem I encountered, because I used the Yaapu lua script on the OpenTX 2.3.9 version before. When using the frsky D16 protocol in the all-in-one tuner built in the TX16s, I can use the s of the Radiomaster R161 receiver. The port's telemetry returns a lot of sensor data from the Pixhawk's Ardupilot firmware and provides it to Yaapu's HUI. Now, I tested various versions from EdgeTX 2.4, 2.5, 2.6RC1-3 and found that none of them solved the problem.

The backhaul telemetry of TBS crsf on each version of edgeTX is normal.

pfeerick, 2022-01-24:
So to be clear:
- Transmitter: TX16S, using MPM
- Receiver: R161
- Telemetry source: Pixhawk Ardupilot, feeding into R161 SPort

And only those five sensors show up on EdgeTX (which look to be the top rows on OpenTX view). 

On interesting thing which I'm not sure of the relevance / importance of is that they don't have an ID number... all the missing ones are '0' ... just thinking that there had better not be something in the code around the telemetry that ignores `0` 😆

richardclli, 2022-01-24:
I think this is the case:

In tememetry_sensors.cpp:

bool TelemetrySensor::isAvailable() const
{
  return ZLEN(label) > 0;
}

And in model_telemetry.cpp:

  for (uint8_t idx = 0, count = 0; idx < MAX_TELEMETRY_SENSORS; idx++) {
    if (g_model.telemetrySensors[idx].isAvailable()) {
      auto txt = new StaticText(window, grid.getLabelSlot(), std::to_string(idx), BUTTON_BACKGROUND, CENTERED | COLOR_THEME_PRIMARY1);
      Button * button = new SensorButton(window, grid.getFieldSlot(), idx, ++count);

Well, for ID = 0, label = NULL and ...... Opts out.

richardclli, 2022-01-24:
When ID is not found in the known list, the ID is a custom ID and the ID will be decoded to become a 4 chars label.  In this case, the isAvailable() should return true.  The problem is not in the display?

pfeerick, 2022-02-07:
Matek F765-WSE which can run Ardupilot arrived today and I have R161 receivers so hopefully I can reproduce this in the next couple of days.

pfeerick, 2022-02-11:
I think I finally have ardupilot and sport telemetry going with the R161 and  F765-WSE ... in addition to the four telemetry elements from the R161 receiver itself, I only get GPS from Ardupilot (i.e. as shown in the OP screenshot) - so hopefully there are more telementry elements coming down the feed and I'm just not see them. Just need to double check on either a B&W radio or possibly OTX and then might have some chance of finding out what is going on.

pfeerick, 2022-02-13:
Right, so with Yappu, and frsky sport (4), you get telemetry sensors, but yappu doesn't seem happy with that by default. 

If I set it to frsky sport (10), you only get five sensors, but yappu is now happy, and if you run the yapuu frsky debug tool you see the data it is looking for is coming in...

[image]
[image]
[image]

With frsky sport (4) the yappu frsky debug tool shows no data.

richardclli, 2023-02-07:
@CNzhangsir  So this can be closed?

gagarinlg, 2023-02-26:
closed, no response

closed 2023-02-26 by gagarinlg