EdgeTX/edgetx #1494: SK and HU translation build failures

pfeerick, 2022-01-25:
@richardclli When doing my multi-language builds for v2.6.0, I hit failures for the SK and HU translations for colorlcd, seemingly because they are missing from here. They are present only as tts_##.cpp files, so I'm not sure what we should do about them... 

This means SK and HU are missing for colorlcd now, and and CN TW have been skipped also since I started doing the builds for 2.5.0

https://github.com/EdgeTX/edgetx/blob/ee21651dda4971600d52668de7d6f79ab1f8afaa/tools/charset.py#L17

```
Traceback (most recent call last):
  File "/src/tools/build-font-bitmap.py", line 153, in <module>
    main()
  File "/src/tools/build-font-bitmap.py", line 148, in main
    font = FontBitmap(args.subset, args.size, args.font, (0, 0, 0), (255, 255, 255))
  File "/src/tools/build-font-bitmap.py", line 15, in __init__
    self.chars = get_chars(language)
  File "/src/tools/charset.py", line 35, in get_chars
    result += "".join([char for char in special_chars[subset]])
KeyError: 'sk'
radio/src/fonts/CMakeFiles/truetype_font_32.dir/build.make:72: recipe for target 'radio/src/fonts/font_32.png' failed
```

```
Job start: 2022-01-24_23-52-00
[ SUCCESS ] nv14-CZ
[ SUCCESS ] nv14-DE
[ SUCCESS ] nv14-EN
[ SUCCESS ] nv14-ES
[ SUCCESS ] nv14-FR
[ SUCCESS ] nv14-IT
[ SUCCESS ] nv14-PT
[    FAIL ] nv14-SK
[ SUCCESS ] nv14-SE
[ SUCCESS ] nv14-PL
[    FAIL ] nv14-HU
[ SUCCESS ] nv14-NL
[ SUCCESS ] t16-CZ
[ SUCCESS ] t16-DE
[ SUCCESS ] t16-EN
[ SUCCESS ] t16-ES
[ SUCCESS ] t16-FR
[ SUCCESS ] t16-IT
[ SUCCESS ] t16-PT
[    FAIL ] t16-SK
[ SUCCESS ] t16-SE
[ SUCCESS ] t16-PL
[    FAIL ] t16-HU
[ SUCCESS ] t16-NL
[ SUCCESS ] t18-CZ
[ SUCCESS ] t18-DE
[ SUCCESS ] t18-EN
[ SUCCESS ] t18-ES
[ SUCCESS ] t18-FR
[ SUCCESS ] t18-IT
[ SUCCESS ] t18-PT
[    FAIL ] t18-SK
[ SUCCESS ] t18-SE
[ SUCCESS ] t18-PL
[    FAIL ] t18-HU
[ SUCCESS ] t18-NL
[ SUCCESS ] t8-CZ
[ SUCCESS ] t8-DE
[ SUCCESS ] t8-EN
[ SUCCESS ] t8-ES
[ SUCCESS ] t8-FR
[ SUCCESS ] t8-IT
[ SUCCESS ] t8-PT
[ SUCCESS ] t8-SK
[ SUCCESS ] t8-SE
[ SUCCESS ] t8-PL
[ SUCCESS ] t8-HU
[ SUCCESS ] t8-NL
[ SUCCESS ] tlite-CZ
[ SUCCESS ] tlite-DE
[ SUCCESS ] tlite-EN
[ SUCCESS ] tlite-ES
[ SUCCESS ] tlite-FR
[ SUCCESS ] tlite-IT
[ SUCCESS ] tlite-PT
[ SUCCESS ] tlite-SK
[ SUCCESS ] tlite-SE
[ SUCCESS ] tlite-PL
[ SUCCESS ] tlite-HU
[ SUCCESS ] tlite-NL
[ SUCCESS ] tx12-CZ
[ SUCCESS ] tx12-DE
[ SUCCESS ] tx12-EN
[ SUCCESS ] tx12-ES
[ SUCCESS ] tx12-FR
[ SUCCESS ] tx12-IT
[ SUCCESS ] tx12-PT
[ SUCCESS ] tx12-SK
[ SUCCESS ] tx12-SE
[ SUCCESS ] tx12-PL
[ SUCCESS ] tx12-HU
[ SUCCESS ] tx12-NL
[ SUCCESS ] tx16s-CZ
[ SUCCESS ] tx16s-DE
[ SUCCESS ] tx16s-EN
[ SUCCESS ] tx16s-ES
[ SUCCESS ] tx16s-FR
[ SUCCESS ] tx16s-IT
[ SUCCESS ] tx16s-PT
[    FAIL ] tx16s-SK
[ SUCCESS ] tx16s-SE
[ SUCCESS ] tx16s-PL
[    FAIL ] tx16s-HU
[ SUCCESS ] tx16s-NL
[ SUCCESS ] x10-CZ
[ SUCCESS ] x10-DE
[ SUCCESS ] x10-EN
[ SUCCESS ] x10-ES
[ SUCCESS ] x10-FR
[ SUCCESS ] x10-IT
[ SUCCESS ] x10-PT
[    FAIL ] x10-SK
[ SUCCESS ] x10-SE
[ SUCCESS ] x10-PL
[    FAIL ] x10-HU
[ SUCCESS ] x10-NL
[ SUCCESS ] x10-access-CZ
[ SUCCESS ] x10-access-DE
[ SUCCESS ] x10-access-EN
[ SUCCESS ] x10-access-ES
[ SUCCESS ] x10-access-FR
[ SUCCESS ] x10-access-IT
[ SUCCESS ] x10-access-PT
[    FAIL ] x10-access-SK
[ SUCCESS ] x10-access-SE
[ SUCCESS ] x10-access-PL
[    FAIL ] x10-access-HU
[ SUCCESS ] x10-access-NL
[ SUCCESS ] x12s-CZ
[ SUCCESS ] x12s-DE
[ SUCCESS ] x12s-EN
[ SUCCESS ] x12s-ES
[ SUCCESS ] x12s-FR
[ SUCCESS ] x12s-IT
[ SUCCESS ] x12s-PT
[    FAIL ] x12s-SK
[ SUCCESS ] x12s-SE
[ SUCCESS ] x12s-PL
[    FAIL ] x12s-HU
[ SUCCESS ] x12s-NL
[ SUCCESS ] x7-CZ
[ SUCCESS ] x7-DE
[ SUCCESS ] x7-EN
[ SUCCESS ] x7-ES
[ SUCCESS ] x7-FR
[ SUCCESS ] x7-IT
[ SUCCESS ] x7-PT
[ SUCCESS ] x7-SK
[ SUCCESS ] x7-SE
[ SUCCESS ] x7-PL
[ SUCCESS ] x7-HU
[ SUCCESS ] x7-NL
[ SUCCESS ] x7-access-CZ
[ SUCCESS ] x7-access-DE
[ SUCCESS ] x7-access-EN
[ SUCCESS ] x7-access-ES
[ SUCCESS ] x7-access-FR
[ SUCCESS ] x7-access-IT
[ SUCCESS ] x7-access-PT
[ SUCCESS ] x7-access-SK
[ SUCCESS ] x7-access-SE
[ SUCCESS ] x7-access-PL
[ SUCCESS ] x7-access-HU
[ SUCCESS ] x7-access-NL
[ SUCCESS ] x9d-CZ
[ SUCCESS ] x9d-DE
[ SUCCESS ] x9d-EN
[ SUCCESS ] x9d-ES
[ SUCCESS ] x9d-FR
[ SUCCESS ] x9d-IT
[ SUCCESS ] x9d-PT
[ SUCCESS ] x9d-SK
[ SUCCESS ] x9d-SE
[ SUCCESS ] x9d-PL
[ SUCCESS ] x9d-HU
[ SUCCESS ] x9d-NL
[ SUCCESS ] x9dp-CZ
[ SUCCESS ] x9dp-DE
[ SUCCESS ] x9dp-EN
[ SUCCESS ] x9dp-ES
[ SUCCESS ] x9dp-FR
[ SUCCESS ] x9dp-IT
[ SUCCESS ] x9dp-PT
[ SUCCESS ] x9dp-SK
[ SUCCESS ] x9dp-SE
[ SUCCESS ] x9dp-PL
[ SUCCESS ] x9dp-HU
[ SUCCESS ] x9dp-NL
[ SUCCESS ] x9dp2019-CZ
[ SUCCESS ] x9dp2019-DE
[ SUCCESS ] x9dp2019-EN
[ SUCCESS ] x9dp2019-ES
[ SUCCESS ] x9dp2019-FR
[ SUCCESS ] x9dp2019-IT
[ SUCCESS ] x9dp2019-PT
[ SUCCESS ] x9dp2019-SK
[ SUCCESS ] x9dp2019-SE
[ SUCCESS ] x9dp2019-PL
[ SUCCESS ] x9dp2019-HU
[ SUCCESS ] x9dp2019-NL
[ SUCCESS ] x9lite-CZ
[ SUCCESS ] x9lite-DE
[ SUCCESS ] x9lite-EN
[ SUCCESS ] x9lite-ES
[ SUCCESS ] x9lite-FR
[ SUCCESS ] x9lite-IT
[ SUCCESS ] x9lite-PT
[ SUCCESS ] x9lite-SK
[ SUCCESS ] x9lite-SE
[ SUCCESS ] x9lite-PL
[ SUCCESS ] x9lite-HU
[ SUCCESS ] x9lite-NL
[ SUCCESS ] x9lites-CZ
[ SUCCESS ] x9lites-DE
[ SUCCESS ] x9lites-EN
[ SUCCESS ] x9lites-ES
[ SUCCESS ] x9lites-FR
[ SUCCESS ] x9lites-IT
[ SUCCESS ] x9lites-PT
[ SUCCESS ] x9lites-SK
[ SUCCESS ] x9lites-SE
[ SUCCESS ] x9lites-PL
[ SUCCESS ] x9lites-HU
[ SUCCESS ] x9lites-NL
[ SUCCESS ] xlite-CZ
[ SUCCESS ] xlite-DE
[ SUCCESS ] xlite-EN
[ SUCCESS ] xlite-ES
[ SUCCESS ] xlite-FR
[ SUCCESS ] xlite-IT
[ SUCCESS ] xlite-PT
[ SUCCESS ] xlite-SK
[ SUCCESS ] xlite-SE
[ SUCCESS ] xlite-PL
[ SUCCESS ] xlite-HU
[ SUCCESS ] xlite-NL
[ SUCCESS ] xlites-CZ
[ SUCCESS ] xlites-DE
[ SUCCESS ] xlites-EN
[ SUCCESS ] xlites-ES
[ SUCCESS ] xlites-FR
[ SUCCESS ] xlites-IT
[ SUCCESS ] xlites-PT
[ SUCCESS ] xlites-SK
[ SUCCESS ] xlites-SE
[ SUCCESS ] xlites-PL
[ SUCCESS ] xlites-HU
[ SUCCESS ] xlites-NL
[ SUCCESS ] tpro-CZ
[ SUCCESS ] tpro-DE
[ SUCCESS ] tpro-EN
[ SUCCESS ] tpro-ES
[ SUCCESS ] tpro-FR
[ SUCCESS ] tpro-IT
[ SUCCESS ] tpro-PT
[ SUCCESS ] tpro-SK
[ SUCCESS ] tpro-SE
[ SUCCESS ] tpro-PL
[ SUCCESS ] tpro-HU
[ SUCCESS ] tpro-NL
[ SUCCESS ] zorro-CZ
[ SUCCESS ] zorro-DE
[ SUCCESS ] zorro-EN
[ SUCCESS ] zorro-ES
[ SUCCESS ] zorro-FR
[ SUCCESS ] zorro-IT
[ SUCCESS ] zorro-PT
[ SUCCESS ] zorro-SK
[ SUCCESS ] zorro-SE
[ SUCCESS ] zorro-PL
[ SUCCESS ] zorro-HU
[ SUCCESS ] zorro-NL
Job end: 2022-01-25_01-09-11
```

richardclli, 2022-01-25:
I think these 2 are never exists in OpenTX 2.4 branch anyway, not only these 2 missing in charset.py, the translation files are not there anyway.  For the .cpp, it is for voices and not for translations.

richardclli, 2022-01-25:
I think we will move and support more translation languages, but we need to work out:
1. Translations reworks
2. LVGL support
3. LVGL fonts and translations mapping

pfeerick, 2022-01-25:
Oh, indeed... hopefully at the end we can use something like https://crowdin.com to offload the translations and make them easier for native speakers to help out. 

So colorlcd just ignore SK and HU for now?

richardclli, 2022-01-25:
Yes, I think these 2 did not exists in OpenTX 2.3 branch either.

richardclli, 2022-01-25:
> Oh, indeed... hopefully at the end we can use something like https://crowdin.com to offload the translations and make them easier for native speakers to help out.
> 
> So colorlcd just ignore SK and HU for now?

We can work out something like crowdin.com if we can manage to work out a freq use subset multi-lang font in LVGL.  And we can do on-the-fly language change and user multi-lang input as well then.

pfeerick, 2022-01-25:
Ah ha, that makes perfect sense now... so don't need to compile for, as they are for the voice language prompt only, not for the display language. Problem solved. Thank you! :)

closed 2022-01-25 by pfeerick