EdgeTX/edgetx #1497: Different Systems on same SDcard

ChrisOhara57, 2022-01-25:
I just did some thinking of how to easily switch between systems (openTx, edgeTx and probably also erskyTx and ETHOS).
I do fly my planes with OTX only but would like do some testing of edgeTx with the same radio, (and a friend of mine does the same with a Horus (openTx and ETHOS).
It is easy to use two different sdcards and flash the radio, I know.
But I do not like the interchange of sdcards on a regular basis.

Is it possible to change the sdcard structure (on long term prospect)
On the first level of the sdcard structure there should be only
/firmware/ (bin files for flashing - all systems)
/edgetx/ (rest of sdcard of edgeTx)
/opentx/ (rest of sdcard of opentx)
/erskytx/ (rest of sdcard of erskytx)
/ethos/ (rest of sdcard of ethos)
Even different versions of one system could be possible (if you do not mind renaming the folder when needed).
I do know that there are different teams involved (so maybe it is not likely to happen).
If you start with edgTx only you could theoretically already store it parallel to one of the others on the same card.
But I do not know whether there are performance issues related to that proposal.

pfeerick, 2022-01-25:
Performance, I wouldn't think that would be an issue... there would just be a base assumption that `/edgetx` would be a prefix folder for all other folders. But there could be a problem with maximum file length as I can't remember if that stacks up. 

As far as the other systems... I don't fancy your chances. When I was switching between OTX and ETX at the start, I just used to rename the RADIO and MODELS folders, as that was all that needed to be protected... everything else is common for them. So I still have a MODELS23 and RADIO23 on my primary TX16S which is the OTX versions of the models it was running at the time.

ChrisOhara57, 2022-01-25:
> 

Do you still use TX16S for both systems (only flashing firmware) when switching between OTX ETX?

pfeerick, 2022-01-25:
Not now, but I was for some time... I would rename the OTX MODEL and RADIO folders RADIO23 and MODELS23, and rename the ETX back to MODELS and RADIO, and then when swtiching from ETX back to OTX rename the ETX MODELS and RADIO to MODELS24 and RADIO24 (as this was when I was using ETX 2.4). I could probably have been a bit smarter and had two scripts in the root of SD card that would have automated that part.

ChrisOhara57, 2022-01-25:
I do have an x9liteS. So I think the two folders would be MODEL (as the folder for the model backup files) and EEPROM (for the backup of the EPROM). (I can not see any RADIO file)
But I do like to use OTX (for flying) and ETX (for tinkering with small/inexpensive models) parallel for a long period.
Also I don't believe that the Sdcard contents will stay compatible for a long time.
So I probably will have to use different SDcards as a long term solution.

pfeerick, 2022-01-26:
Since it's a B&W radio, you would have to backup your EEPROM, which should contain both model and radio data. Assuming no changes to the model, using ETX 2.5, you could just switch between EEPROM and FIRMWARE via the bootloader. If you use ETX 2.6, it won't use the EEPROM - it uses the SDCard instead so you'll have MODELS and RADIO folders with yml files in them. I've not tried byself... but that could be the perfect way to use OTX and ETX on B&W radios for now... as OTX would use the EEPROM for settings, and ETX would use the SD card.

gagarinlg, 2023-03-18:
I don't think that this is going to be implemented. I will close this issue soon, when nobody objects

closed 2023-04-07 by pfeerick