EdgeTX/edgetx #1500: Negative GVs are messed up in YAML

jfrickmann, 2022-01-25:
In the attached .etx file I have a model where:
- GV1 is "GVA"
- GV2 is "GVB"

The four first channels each have the following weights in mixes:
1. GVA
2. -GVA
3. GVB
4. -GVB

But when I simulate (and YAML is generated and read by the simulator), I get:
1. GVA
2. GVA
3. GVB
4. -GVA

As you can see, both of the cases with negative sign are incorrect. I discovered the issue when writing models to my Tx after loading 2.6. It looks like it is all the same for TX16S and X9D+ (only verified in simulator). I noticed the issue where GVs were used for adding differential, but it is the same here for weights.
[YAML-test.zip](https://github.com/EdgeTX/edgetx/files/7937151/YAML-test.zip)

pfeerick, 2022-01-26:
Looks like companion is generating the correct code

```
mixData:
  - destCh: 0
    srcRaw: I0
    weight: GV1
...
  - destCh: 1
    srcRaw: I1
    weight: -GV1
...
  - destCh: 2
    srcRaw: I2
    weight: GV2
...
  - destCh: 3
    srcRaw: I3
    weight: -GV2
```

but the radios isn't, and thus probably also why it's reading it wrong...
```
mixData: 
 -
   weight: GV1
   destCh: 0
   srcRaw: I0
...
 -
   weight: -GV2
   destCh: 1
   srcRaw: I1
...
 -
   weight: GV2
   destCh: 2
   srcRaw: I2
...
 -
   weight: -GV3
   destCh: 3
   srcRaw: I3
```

jfrickmann, 2022-01-27:
I have now verified that on the radio, the GVs are used as displayed. I.e. it is not just an issue with the screen displaying the wron values.
If I read data back into Companion 2.6, it is corrected again. I.e. the radio is reading and writing the negative GVs differently from Companion.

jfrickmann, 2022-01-27:
I think that I fixed it in #1516

jfrickmann, 2022-01-27:
Oops, there is also a problem when the radio writes back data!

jfrickmann, 2022-01-27:
Should be fixed now

closed 2022-02-01 by pfeerick