EdgeTX/edgetx #1503: power on delay is not working. stays at 3 seconds.

P-I-Engineer, 2022-01-26:
**Describe the bug**
power on delay is not working. set to 0s or 1s, power on delay is still taking about 3 seconds.
power off delay appears to be working.

**To Reproduce**
have to hold power button for 3 seconds to turn on 

**Expected behavior**
should turn off and turn on with quick click

**Screenshots**

**General information**
2.6
x9-lite

dailrus, 2022-02-03:
Same situation on Jumper T12

Directory, 2022-02-04:
same on qx7. imo this is a serious bug and santa should have never released without this being fixed. this actively ruins the user experience from what it was before.

wang322588, 2022-02-05:
Same situation on Jumper T-lite

pfeerick, 2022-03-03:
I think I finally may have found the cause of this - thanks to https://github.com/EdgeTX/edgetx/issues/1366#issuecomment-1021161968 - as I got my TX12 to exhibit this exact same behavior. Can you check in your `RADIO/radio.yml` file, and see if the `version:` parameter is 220 still - it should be 221. If so, once I find ou why the updated version tag isn't being written this can be put to rest.

Directory, 2022-03-03:
the version is indeed 220. and making it 221 did fix this sort of. its still slower then it used to be. but the dot delay is gone (since my power on delay is 0s). but before it would instantly power on from the slightest click of the power button. now i need to still hold it for a brief moment before it registers. still faster then the 3-4 seconds before. i think the delay is simply a side effect of having to read the setting from a slow sd card rather than eeprom.

closed 2022-03-20 by pfeerick