EdgeTX/edgetx #1504: 2.6.0 Switch Warnings on Pots

techfixpros, 2022-01-26:
eTx 2.6.0 b338d4d2
Taranis X9D

On bootup, switch warning pops up and no switch/pots is shown when all switches are in correct position and Pot Warning:Manual is enabled. Disabling Pot Warning corrects issue. 

With Pot Warning set to Manual and warning is shown (without indicating which pot is in the wrong position), toggling a switch will show the switch position, and S1 - S2 being in the incorrect position. Cycling through all pot positions does not clear either pot from the warning state. Toggling switch to default position clears the warning indication for the switch, and S1 - S2 disappear without clearing the warning prompt.

Functioned properly in 2.5.0.

dududu76, 2022-03-01:
Additionnally, the status of pots warnings are not correctly read/restored when backed-up to computer using companion.

elecpower, 2022-05-20:
From X9D+ libsim

[image]

PR #1981 (2022-06-07): Fix pot and slider warnings in Companion and radios
Fixes #1843
Fixes #1730
Fixes #1663
Fixes #1504
Fixes #1050

Summary of changes:
- fix Companion yaml model read and write
- B&W radios treat bit values the same as color to avoid unnecessary complication for Companion and radio conversions
- B&W radios default is therefore off for all pots and sliders
- B&W radios fix runtime warnings check
- all radios fix to save last pot or slider position when checked and mode AUTO on change of model or power off
- all radios do not display Pots Warning in model setup if no pots or sliders supported

Not [... cut]
closed 2022-06-07 by pfeerick