EdgeTX/edgetx #1505: 2.6.0 automatic turn on when turn off and release power button after few seconds

chan2328, 2022-01-26:
previously I used official 2.5.0, there is no problem with turning off. but updated to official 2.6.0 it will automatically turn on when turn off and release power button after 2-3 seconds.

* I am using chinese version flysky NV14 nirvana. I did not do any mod.

pfeerick, 2022-01-26:
That sounds more like some button bounce... as early in 2.6.0 we basically removed the power-on delay - i.e. you can press and hold the buttons for a quarter of a second, and release, and it will turn on, even though screen is still black. 

If you release one button first, and then the other, when turning off, does it still do this? Maybe we need to re-add at least a partial power on delay to stop that.

chan2328, 2022-01-26:
The problem is now that I cannot turn off the radio in 2.6.0. by pressing the power button, but I can turn off in 2.5.0.
When I pressed the power button to turn off the radio, the screen is going to black and screen light is off, but when I release the power button, the screen light will turn on immediately, and the radio will turn on after few second.

discord5000, 2022-01-26:
I can't reproduce

discord5000, 2022-01-26:
Are you getting the haptic vibrate before you release the buttons?

chan2328, 2022-01-26:
> Are you getting the haptic vibrate before you release the buttons?

the motor vibrate after power off timer, and before the screen is going to black

chan2328, 2022-01-26:
here is a video show the problem
https://www.youtube.com/watch?v=im_3LLxp8vo

chan2328, 2022-01-26:
Now, every time I Turn off the radio by pulling out the battery out, and then press power button to turn on the radio, it goes to emergency mode. after that press the power button again, it will go to normal opening screen.

tfurci, 2022-02-01:
Happend to me to on TX16S. reseated ribbon cables and RTC battery. Problem seems to be fixed now

RIPNDIP-FPV, 2022-05-30:
Currently experiencing this with a brand new NV14 flashed to 2.6.0, 2.7.1, and the nightly build. Batteries have to be completely removed to shut the radio down. It does not turn itself back on once the batteries are reinserted, but it does tend to go into emergency mode after shutting down in this manner.
Can confirm the issue does not occur on EdgeTX version 2.5.0

pfeerick, 2022-05-31:
Removing the batteries while a radio is operating will *always* result in a "emergency mode" afterwards - this is the radios way ot telling you "either I crashed and just rebooted, or something weird is going on with the power" - it is supposed to be functioning in a reduced capability mode that ensures you still have control of the active model - nothing more or less. 

I think around 2.6.0 we reduced the power on detection time... think it's around 100ms now... maybe there is some button debounce or something happening and we need to wind it back up again. I can't say I've noticed this myself on the NV14. But I also have more of a habit of releasing one button at a time rather than both together (which _could_ reduce switch bounce).

pfeerick, 2022-11-24:
@chan2328 Is this still an issue?

RIPNDIP-FPV, 2022-11-25:
Will test later tonight.

closed 2023-09-05 by raphaelcoeffic

darshanmahakul, 2024-01-30:
i have also facing the same issues that when turning off the transmitter it again turns on and when 1st time inserting battery into it, it showing emergency mode

pfeerick, 2024-01-31:
Please make sure you have flashed both the bootloader and main firmware with a recent version (i.e. 2.9.x) and see if this still happens.

darshanmahakul, 2024-01-31:
bootloader is edgetx buddy

pfeerick, 2024-02-01:
If you're using EdgeTX Buddy and flashing using DFU mode, it flashes both the bootloader and the firmware in one step (the bootloader is what runs when you hold the trims inwards and power on the radio 😉). 

I'm sorry but I can't reproduce this on either the NV14 or the EL18, so it may be something specific to your hardware. Have any mods been done to it, such as the power mods for crossfire modules, etc?