EdgeTX/edgetx #1506: Add label to receiver number

Combinacijus, 2022-01-26:
## Problem
In Model Setup there is an option to bind receiver with specific Receiver Number (RxNum) 0-63 this is a nice feature but I see problems with multiple receivers:
- How do you even remember which number is bound to which receiver especially if not used often. 
- What if you setup RxNum not in order (01, 02 for drones, 11, 12 for planes etc.) how do you easily know which numbers are used and which one to select.

## Proposed solutions
- Simplest solution would be to add a new editable label field next to RxNum. Therefore instead of number you could name receiver like "Racer1 ERLS", "Freestyle HD TBS" etc. instead of 01, 02.
[image]

- I think that a more complete solution would be to make a menu list of RxNum and labels either like Mixes tab that shows all available numbers and labels or even better if the list is appendable and only used entries are show to reduce clutter.

Combinacijus, 2022-01-26:
First solution looks simple enough I might even try to implement it myself on the weekend if I'll manage to setup development environment

ParkerEde, 2022-01-26:
I always use the file number of modelxx.yml as receiver number. So i am sure, it is unique.
What do you think about it to suggest the modelxx.yml number as receivernumber when the model is new created or copied?

Combinacijus, 2022-01-26:
@ParkerEde That's a good workaround on current implementation.
Not sure I understood the second part tho. Do you mean for example when you create a new model e.g. model05.yml then automatically assign 05 to RxNum? In that case it wouldn't make sure it's unique if in previous models you assigned 05. Also multiple models could have same RxNum
In case of copy I think it should use same RxNum

pfeerick, 2022-01-26:
What we could possibly do now since we have the global `models.yml` is include the receiver ids there, and thus have a single place where they can be scanned for, and the next free RxNum allocated by default. This also means this is how the firmware can identify what models have that ID number in use, and display it. Alternately, if we don't want to clutter that file, a separate file - but I'm not sure that is warranted. 

What I think ParkerEde was suggesting was since the number in `modelxx.yml` is itself unique, if you don't go changing the numbers manually, then it would a good source for the number... but I think it is better to have a single reference point and keep updating it. And yes, a copy of a model should not change anything (other than the filename), thereby should have a RxNum collision.

dlktdr, 2022-01-27:
If/when we encorporate #1374 I already have the labels.yml saving the receiver id and modified the isunique calls. Could pretty easily add a label to it for reference if that's a good idea too. GUI changes to check if it's been used or not haven't been done yet.

pfeerick, 2022-01-27:
**when, not if** ;)

dlktdr, 2022-01-28:
Added a warning when you pick a receiver id is used.. Needs a bit of love on the 
[image]

Needs a bit of love auto-updating in the main window still but thinking showing it here is a good idea too.
[image]

pfeerick, 2022-01-28:
Damn, that was fast! :)

gagarinlg, 2023-03-18:
This is implemented 2.8

closed 2023-03-18 by gagarinlg