EdgeTX/edgetx #1507: LUAC Filter in SD Card Explorer

ParkerEde, 2022-01-26:
If I explore the SD-Card for example in "SCRIPTS" folder to start a lua script, then I would like to see all lua files *.lua but not the *.luac files. Is it possible, to filter them generally away?

pfeerick, 2022-01-27:
That needs a little thinking about how to make that usable, as the sd card browser currently operates as you would expect - it shows all files on the card, regardless of if it knows what to do with them or not. 

What I think you're really after there is a Lua script browser...

ParkerEde, 2022-01-27:
the display of the luac files makes it very confusing to start the right file, if you have several lua files in the script folder. 
What do you think about a switch in the mainmenu: "hide luac files"?

piotrva, 2022-02-07:
Yup I also find it confusing that .luac files are being displayed.
And yup, I think additional menu for displaying only .lua files would be better the globally filtering .luac files in SD browser.

piotrva, 2022-02-07:
@JimB40 That is also the point - so the feature should:

- If only XYZ.luac file is present - show luac file
- If both XYZ.lua and XYZ.luac are present - if .lua is newer than .luac - run .lua that will generate .luac and set .luac file date to bigger value of [present radio date, date of creation of .lua +"1"]. If .luac file is newer - run luac.

So the "Lua Browser" should in my opinion hide file extension and display "scripts" without extension using algorithm presented above.

What do you think?

pfeerick, 2022-02-08:
This all sounds good (truly!), but how practical is this going to be/how will it look? Do we make it so the SD browser actually does have options (if so, what other options would be useful), or do we add yet another viewer? Or is this better as a lua app itself, which you launch from Tools?

JimB40, 2022-02-08:
Points to consider
1. This is not "LUA browser" but "SD Card browser" so shoud stay generic
2. Memory management is not perfect so on BW targets compilation takes away some memory. Read: it happens that if you compile lots of lua (like BF LUA does) it radio may threow error due to lack of mem. (until you power recycle). That's why some more complex apps are distributed with *.luac
AFAIR iNav does the trick that files have *.lua extension but in fact they are compiled versions.
  
As far as I understand idea to declutter I use SD Card browser to check if there ARE compiled versions :)
So if we go this path it shoud be an option.

piotrva, 2022-02-08:
@JimB40:
1. I was thinking about SEPARATE menu for only browsing lua files, not affecting existing SD Browser.
2. That is a point to consider