EdgeTX/edgetx #1509: timer hesitates and skips numbers.

Daannoa, 2022-01-26:
Hello,
The timer seems to have a problem: the timer hesitates and skips numbers.
I have the same problem as shown in this video:
[https://www.youtube.com/watch?v=e-PmbOXriws](url)
The problem appears on the two Radiomaster TX16S that I tested.

**General information**
 - EdgeTX version 2.6.0 
 - Hardware: Radiomaster TX16S

pfeerick, 2022-01-26:
This isn't specific to 2.6 btw - it's always been there... even back in the OTX days IIRC. I believe it is because Timer2 is a Lua widget, and is a result of the amount of CPU time it is allowed -> how many times it can check if 1s has passed, and then how long it takes to determine what to do and update the screen. This may coincidentally be fixed by a PR that aims to give more resources to widgets - https://github.com/EdgeTX/edgetx/pull/1498

Daannoa, 2022-04-17:
Hello,
I allow myself to come back to make a return on this point after my upgrade to Edgetx 2.7.
Unfortunately the situation remains the same: the timer hesitates and skips numbers.

pfeerick, 2022-04-17:
As the poster of that video indicated in a comment, and as I said above... it is an issue with the Timer 2 lua - if you run the same version of the Lua on OpenTX it skips there also. The widget is updating fast enough, still like 10 times a second, but it looks like the math it using is *too* precise a floating point number, and isn't being rounded, hence the skip.

@offer-shmuely Any chance you can sort that out? It looks like the dd_raw value needs to be trimmed and rounded off?

offer-shmuely, 2022-04-17:
Of course!
Currently on vacation, but later this week I will check it.
I do assume that adding more resources to widget, shouldn’t be the solution in this case anyway,
This is simple widget, that should not consume a lot of cpu,
Let’s keep the “a lot of cpu” to the real fancy widgets (one of them will com soon… )

pfeerick, 2022-04-18:
Thank you... and enjoy your vacation! 🙂 

Yeah, it's not a resource problem... the widget is refreshing plenty fast enough, but the floating point math is tricking it into showing the a digit for too long, thus the jumps in number shown. I will also try to look at it, but my Lua isn't the best, so can't promist anything :)

offer-shmuely, 2022-04-18:
fixed in pull-request: /EdgeTX/edgetx-sdcard/pull/59

pfeerick, 2022-04-19:
Works perfectly... thanks for that! :)

As a reference, the widget above timer 2 is the counter widget,  and is showing widget update speed...
https://user-images.githubusercontent.com/5500713/163896408-9fa7779e-8afd-4079-853c-2aced44cc381.mp4

@Daannoa This means you need to update your sdcard pack with the latest release @ https://github.com/EdgeTX/edgetx-sdcard/releases/tag/latest OR replace `WIDGETS/Timer2/main.lua` on your SD card with [this file](https://raw.githubusercontent.com/EdgeTX/edgetx-sdcard/master/sdcard/horus/WIDGETS/Timer2/main.lua). (right click and save)

closed 2022-04-19 by pfeerick

offer-shmuely, 2022-04-19:
why do we need a counter widget on non developer transmitter?
can we remove if from the git?

pfeerick, 2022-04-19:
Probably... it has been there forever and I haven't seen any complaints now though... :shrug: I'll remember this post though for the inevitable *where did the counter widget go... I use it!* complaint that will probably arise from someone when it is removed 😆

Daannoa, 2022-04-20:
Hello.

First of all, a big thank you for this modification / update of the lua script.

I also want to apologize if I sounded like I was pushing for this update.
I assumed (wrongly) this update will come with the release of EdgeTX2.7.0, so I just wanted to provide feedback.

Thanks again for your amazing work.

offer-shmuely, 2022-04-20:
not "pushing"..... just "reminding" ;-)
did you check it, does it solve the issue on you transmitter as well?