EdgeTX/edgetx #1510: Ghost 12 bit - Companion support needed

pfeerick, 2022-01-27:
With the introduction of #1233 & #1377, Companion needs to be taught about the new Ghost module setting:
```
moduleData: 
   1:
      type: TYPE_GHOST
      subType: 0
      channelsStart: 0
      channelsCount: 16
      failsafeMode: NOT_SET
      mod: 
         ghost: 
            raw12bits: 1
```

elecpower, 2022-01-30:
I'm going back to working on the modules area for MPM so I'll take it if you have no objection

pfeerick, 2022-01-30:
Of course not! You're the Companion whisperer ;) Issue is here so either *someone* can pick it up, and to remind me if I want to do it ;)

TheIsotopes, 2022-02-13:
Any news about this issue?
Both PR #1233 & #1377 was merged but no Companion support for the 12bit mode setting.

commit 00c5e4dd (2022-03-13): chore: Companion - add support for 12bit GHOST opt
Resolves #1510
commit 4571c7b7 (2022-03-16): chore: Companion - add support for GHOST 12bit opt (#1689)

* chore: Companion - add support for 12bit GHOST opt
Resolves #1510

* Ghost protocol reading and writing non-yaml formats