EdgeTX/edgetx #1511: Radiomaster TX12 Random Power Off

baconninja87, 2022-01-27:
I flashed from the stock version of OpenTX on my RM TX12 to EdgeTX 2.6.  I carried nothing over from my old firmware to the new and flashed ETX via SD card with a brand new name brand card.  After flashing, my radio will now shut off at random intervals ranging from 15 seconds to 5 minutes.  This has happened repeatedly 20 times in the first 45 minutes after flashing.  I have tried removing my ELRS module and running with neither the internal or external bay activated, running just the ELRS module in 250 and 500hz modes, running just the internal module, running in USB joystick mode, and running without USB connected.  I also tried to re-flash via companion with the same results afterwards.  The radio seems to function fine right up until the random power off.

baconninja87, 2022-01-29:
Tried a DFU flash again just for the fun of it.  Still randomly powers off.  I did notice the high voltage limit was 8.0 by default so i set that to 8.4, but it still continued to have an issue.  I took it out today to see if it would do it while in use.  Two fail-safes on elrs 250hz shows no change thus far.

piotrva, 2022-02-06:
Is it doing the same with OpenTX?
Is it a new one or used one?
What kind of cells you are using for power?

baconninja87, 2022-02-06:
It does not do the same when running OpenTX.
The radio was new as of 8 months ago.
I am using Sony 18650's in it.

piotrva, 2022-02-06:
Could you please specify exact firmware you are using at the moment? (Go to s SYS -> Version <page 7/7> and please note everything in line VERS: ....)

closed 2023-09-05 by raphaelcoeffic