EdgeTX/edgetx #1515: 2.6.0 theme editor going into emergency mode when saving

rburrow87, 2022-01-27:
TX16S 2.6.0 release

After editing a theme, the transmitter goes into emergency mode when you exit out and save.

pfeerick, 2022-01-28:
Can you give any more guidance as to how to reproduce this, as while this is the second report of this I've seen now, I've having trouble producing it.  :-/ I thought there was some possible connection with having made a new theme (duplicate and edit?, on the second edit?). I've gone through, duplicated a couple of themes, edited them several times and it is behaving.

coaster007, 2022-01-28:
Just tried on linux with simulator27 (last compiled yesterday)

1. goto sys->theme editor
2. press edit theme

core dump last entries:
3,28s: f_open(/home/grazi/Documents/EachineSD/SOUNDS/en/fmnorm.wav, 1) = 0x7f85180030b0 (FIL 0x7f853e15be60)
3,28s: scan [255]
3,29s: scan [255]
3,29s: scan [255]
3,30s: scan [255]
3,30s: scan [255]
3,31s: scan [255]
3,31s: scan [255]
3,31s: proto scan timeout!
3,32s: scan [255]
4,93s: f_close(0x7f85180030b0) (FIL:0x7f853e15be60)
4,94s: f_open(/home/grazi/Documents/EachineSD/SOUNDS/en/stoff.wav, 1) = 0x7f85180030b0 (FIL 0x7f853e15be60)
6,23s: f_close(0x7f85180030b0) (FIL:0x7f853e15be60)
7,04s: f_open(/home/grazi/Documents/EachineSD/THEMES/BurgundyRed/logo.png, 1) = 0x7f850005f6e0 (FIL 0x7f853e15d400)
7,04s: f_close(0x7f850005f6e0) (FIL:0x7f853e15d400)
7,04s: f_open(/home/grazi/Documents/EachineSD/THEMES/BurgundyRed/logo.png, 1) = 0x7f850005f6e0 (FIL 0x7f853e15d400)
7,07s: f_close(0x7f850005f6e0) (FIL:0x7f853e15d400)
*** buffer overflow detected ***: terminated
Aborted (core dumped)

rburrow87, 2022-01-28:
Okay, I tried replicating it again this morning and it wasn't crashing at first, but I figured out how to make it repeatable. The TX needs to be powered off and back on before it starts happening, and it needs to be the active theme.

1. Pick a theme to edit (or duplicate one and edit that)
2. Change something in the theme (if this was the first edit after selecting the theme, it may not go into EM)
3. Power down radio
4. Power up radio
5. Edit the current theme and it should go into EM when you back out and save

At least that's what's needed for it to happen on the radio. I'm not sure how the one in the simulator is doing it too if it requires a power cycle, but I've never really messed with the sim much either.

pfeerick, 2022-01-29:
Right, had to go around several times, but it finally triggered, so one step closer... Thanks!! @rburrow87 

@coaster007 That means https://github.com/EdgeTX/edgetx/issues/1280 is still around, and the two are probably related. I had issues with `simu`, simulator27s little sibling, crashing when trying to go into the theme editor during it's development stage, and that went away when I reset my themes directory, so I think there is something weird going on in relation to the combinations of themes and files present.

kevinkoenig, 2022-01-29:
Can I help?

pfeerick, 2022-01-29:
Hopefully yes... I'll hopefully be able to plug the debugger into the radio tonight... actually, the NV14 should suffer form this as well.. if so... it's already hooked up waiting for some debugging to happen... anyway, I digress... if I can pin it to a particular section of code, I'll let you know where to start looking ;) Also we need to move themer to using built in yaml routines... I think?

pfeerick, 2022-01-29:
After going around, and around and around several times, making incremental edits to various bits of a theme on the NV14... finally got it to splat... still says 'save theme' since it's halted on the debugger....

[image]

```
Level,Function,Stack Frame,Source,PC,Return Address,Stack Used
0,<HardFault Exception>,104 @ 0x20009068,,0x080A70E8,[0x20009080]: 0x0802DCC0,808
0,Window::invalidate, 24 @ 0x200090D0,window.h:368:7,0x0802DCBE,[0x200090E4]: 0x0808C5BA,704
0,StaticText::setText, 32 @ 0x200090E8,static.h:50:9,0x0808C5B6,[0x20009104]: 0x0808CD70,680
0,ThemeSetupPage::setAuthor,296 @ 0x20009108,new_allocator.h:79:7,0x0808CD6C,[0x2000922C]: 0x0808E2F8,648
0,_Function_handler<void, 56 @ 0x20009230,radio_theme.cpp:527:17,0x0808E2F4,[0x20009264]: 0x0808C33E,352
0,ThemeEditPage::deleteLater, 40 @ 0x20009268,std_function.h:248:29,0x0808C33C,[0x2000928C]: 0x0806085E,296
0,_Function_handler<unsigned char,  0 @ 0x20009290,page.cpp:33:10,0x0806085C,<inline>: 0x0806085E,256
0,_Function_handler<unsigned char,  8 @ 0x20009290,invoke.h:59:5,0x08060852,[0x20009294]: 0x08036758,256
0,Button::onPress,  8 @ 0x20009298,std_function.h:248:29,0x08036756,[0x2000929C]: 0x080367CA,248
0,Button::onTouchEnd,  8 @ 0x200092A0,button.cpp:66:5,0x080367C6,[0x200092A4]: 0x080359B6,240
0,Window::forwardTouchEnd,  0 @ 0x200092A8,window.cpp:435:7,0x080359B4,<inline>: 0x080359C4,232
0,Window::forwardTouchEnd, 24 @ 0x200092A8,window.cpp:429:5,0x08035978,[0x200092BC]: 0x080359CC,232
0,Window::onTouchEnd,  8 @ 0x200092C0,window.cpp:446:3,0x080359C8,[0x200092C4]: 0x080359B6,208
0,Window::forwardTouchEnd,  0 @ 0x200092C8,window.cpp:435:7,0x080359B4,<inline>: 0x080359C4,200
0,Window::forwardTouchEnd, 24 @ 0x200092C8,window.cpp:429:5,0x08035978,[0x200092DC]: 0x080359CC,200
0,Window::onTouchEnd,  8 @ 0x200092E0,window.cpp:446:3,0x080359C8,[0x200092E4]: 0x080608EA,176
0,Page::onTouchEnd, 24 @ 0x200092E8,page.cpp:94:3,0x080608E6,[0x200092FC]: 0x080359B6,168
0,Window::forwardTouchEnd,  0 @ 0x20009300,window.cpp:435:7,0x080359B4,<inline>: 0x080359C4,144
0,Window::forwardTouchEnd, 24 @ 0x20009300,window.cpp:429:5,0x08035978,[0x20009314]: 0x080359CC,144
0,Window::onTouchEnd,  8 @ 0x20009318,window.cpp:446:3,0x080359C8,[0x2000931C]: 0x08036D90,120
0,MainWindow::checkEvents, 64 @ 0x20009320,mainwindow.cpp:131:5,0x08036D8E,[0x2000935C]: 0x08036F54,112
0,MainWindow::run, 16 @ 0x20009360,mainwindow.cpp:215:3,0x08036F52,[0x2000936C]: 0x08029C9E, 48
0,guiMain, 16 @ 0x20009370,main.cpp:380:3,0x08029C9A,[0x2000937C]: 0x0802A0DA, 32
0,menusTask, 16 @ 0x20009380,tasks.cpp:254:5,0x0802A0D6,[0x2000938C]: 0x080A9130, 16
0,vPortEnableVFP,  0 @ 0x20009390,port.c:714:5,0x080A912C,R14: 0xFFFFFFFF,  0
0,Top of stack - invalid return address location: R14,,,,,
```

rotorman, 2022-01-29:
The Ozone shot above sounds similar to the issue we had with @raphaelcoeffic some 1-2 weeks ago with Zorro, DEBUG and CLI, where the issue was that audio stack overrun (by increasing audio stack from 400 to 512 double words, the issue went away).
One could check the stack with uxTaskGetStackHighWaterMark() or put an assert in vApplicationStackOverflowHook().
https://www.freertos.org/Stacks-and-stack-overflow-checking.html
The comments on the following lines need to be swapped for this:
https://github.com/EdgeTX/edgetx/blob/a052ba63c2fb1c6d74d06d9ca27a8efb3eb0d633/radio/src/FreeRTOSConfig.h#L54-L55

nostromo123, 2022-01-29:
I see this issue (emergency mode) regular when trying to change a name and save it in this window on my X10 Express. Had it on 2.5 and now also on 2.6.
[image]

joneszee69, 2022-04-02:
I am also having this problem on my 16s radio (emergency mode). Don’t believe I had it on the earlier version (2.5) but seeing it on the 2.6 update. I am noticing it under different circumstances, for instance when I went to start the radio up one time It just started up in emergency mode. Then in some cases why I was just going through different screens it just happens and have to shut off radio and reboot. I am going to pay more attention to when it happens so I could get back to you guys with a better and more accurate information as to when and how it happens exactly.

pfeerick, 2022-04-02:
Going into emergency mode right from startup suggests something went wrong in the last session... battery disconnect. 

But for anything else, if you can get something half reproduable we will have a better chance of getting rid of this bug. I need to come back to this with the debugger to see if I can follow the rabbit down the hole further :grin:

joneszee69, 2022-04-13:
> I will certainly pay better attention and log any and all findings I come up with and will try to get a screenshot also
>
> Thank you and have a great day!!!
> Sent from my iPhoneX

OK so now I am getting the emergency Mode when I try to use USB to computer to make changes on my SD card after a few minutes radio shuts off by itself and when I turn it back on it starts up in emergency mode until I power down again. Also a new problem I’m having is when I power my radio on initially it takes about a minute before the screen load/turn on. It used to turn on within seconds. Not sure if this has anything to do with anything. Thanks

Thank you and have a great day!!!

gagarinlg, 2022-09-17:
I can not reproduce this with the current main. Whenthere are co compaints within this weekend I will close this issue. The Model categories do not exist anymore and the theme editor got an overhaul.

closed 2022-09-22 by gagarinlg

joneszee69, 2022-10-11:
I will certainly pay better attention and log any and all findings I come up with and will try to get a screenshot also

Thank you and have a great day!!!