EdgeTX/edgetx #1517: Companion 2.6 Crashing

ammikey, 2022-01-28:
Companion 2.6 on a windows 10 computer running intel i5
Radio Master TX16S with latest firmware. 
A new theme is created in the radio tested and working correctly. The new theme file is then saved to the sd card contents 
on the PC Which companion uses. Companion will no longer start. If the theme file is removed from the themes folder
Companion will once again start with no issues.

elecpower, 2022-01-30:
@ammikey how are you starting Companion? Program icon or models and settings etx file? 
If program icon, does it crash before or after opening a models and settings etx file?

ammikey, 2022-01-30:
Hi, I need to clarify the problem. Companion will open and I can open the most recent saved radio file. The crash occurs when I try to open a saved aircraft file and there is a custom theme file saved in the themes folder. Companion then becomes unresponsive and crashes.

ammikey, 2022-01-31:
Problem solved. I was not naming the the background file attached to the custom theme incorrectly. Apparently Companion could not determine which background file to use when opening an aircraft file and would close. After naming the background files correctly, (background_480x272.png) all the custom themes are working perfectly.

elecpower, 2022-01-31:
Thank you for the update. Companion should be able to handle gracefully.

elecpower, 2022-02-06:
I have renamed theme files so they do not exit and cannot get Companion to crash in Ubuntu 20.04. The code checks for the existence of file before trying to open and display.
@ammikey can you provide any more information?

PR #1970 (2022-06-04): Stop Companion crashing due to malformed theme.yml files
Fixes #1961
Fixes #1517

Summary of changes:
- trap and report YAML parsing errors when processing theme.yml files
- continuing loading model

Example
[image]
closed 2022-06-04 by pfeerick