EdgeTX/edgetx #1519: Companion V2.6.0 does not show switchnames in Logical Switches folder

Kreisflieger, 2022-01-28:
Loading a model from a Radiomaster TX16S via EdgeTX-Companion 2.6.0. the model is correctly loaded into Companion but the switch names in the Logical Switch folder are not shown. see enclosed picture.
 
[image]

[image]

Kind regards

Stephan

nostromo123, 2022-01-29:
edgeTX 2.6 here

[image]

elecpower, 2022-01-29:
Can you please check Companion radio settings hardware tab for space(s) in switch name. I can replicate if a space.

Kreisflieger, 2022-01-30:
I provide hereby the stored .etx file for further investigation.

Stephan

[image]
.

Kreisflieger, 2022-01-30:
Dear Neil,

 

I can provide the stored Windows 10 .etx file for further investigation.

 

Kind regards and thanks a lot for your fantastic work.

 

Stephan

Logical Switches folder (Issue #1519)

 

<https://github.com/EdgeTX/edgetx/issues/1519#issuecomment-1025002214> , or
unsubscribe
RWB7ANCNFSM5NA2DXCA> .
t=8&pt=524675>  or Android
tm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>
<https://github.com/notifications/beacon/AVZNLOKGNXGWBSGPLGH2HODUYRWB7A5CNFS
M5NA2DXCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOHUMEVZQ

closed 2022-01-30 by Kreisflieger

pfeerick, 2022-01-30:
Perfect... it looks like it was most likely the culprit, so #1524 should fix it as the names _are_ all spaces for whatever reason...

[image]

elecpower, 2022-01-30:
I do not remember seeing quoted spaces before...

pfeerick, 2022-01-31:
Normally it wouldn't be... i.e default on TX16s if I haven't assigned any names is:

```
   SE:
      type: 3pos
      name: 
   SF:
      type: 2pos
      name: 
   SG:
      type: 3pos
      name: 
```

I'm going to assume in this instance it's some weird conversion gremlin...

PR #1524 (2022-01-30): Companion use default hardware name when name only spaces
Fixes #1519