EdgeTX/edgetx #1520: Missing freq. fine tune

sharonx81, 2022-01-28:
Hi. I upgraded my RadioMaster x16S max to EdgeTX 2.6,and although i absolutely love it, it has a problem. The radio frequency fine tune option is gone. it was there on the OprnTX.
i need it to bind my tx to JJRC H36 quad (the freq. tune must be set to 2 for it to work).
by the way, The multi receiver is updated to the latest version

pfeerick, 2022-01-29:
So is this using the `E010R5` protocol?

Are you sure it is the freqency tune you need to adjust, and not the model id? Frequency tuning is only required for the CC2500 RF (i.e. for 1 of the 4 RF chips used on the 4in1 MPM). and the E010R5 protocol is handled by the CYRF6936, not the CC2500, and hence is not listed as a protocol that [requires frequency tuning](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md). It should only be shown on the radio interface if the MPM specifically indicates that option is required (or if there is a bug, naturally).

Flierman1945, 2022-02-01:
Some of the models you can select from the bind list do not seem to have tx fine tune available.  But as a test  set to fr-sky and select any of their rx, the fine tune will be there.

gagarinlg, 2023-03-18:
Only those protocols using one specific RF (CC2500?) chip need this.

closed 2023-03-18 by gagarinlg