EdgeTX/edgetx #1521: Reset to default for Yaml - particuarly for B&W

pfeerick, 2022-01-29:
### Discussed in https://github.com/EdgeTX/edgetx/discussions/1514

<div type='discussions-op-text'>

<sup>Originally posted by **Denethor85** January 28, 2022</sup>
in the new version 2.6.0 there is no eeprom reset button, so how can I reset each setting to default?  I tried reinstalling the firmware, wiping the SD card, but none of this seems to work ....

Thanks</div>

Essentially, we need to brink back the reset to factory option, and then have some way to prevent a import of EEPROM on the reset. Perhaps a empty file that indicates clean slate is wanted, and is then deleted?

stephenaa, 2022-02-12:
Where would the UI of the "Reset to factory defaults" be?

stephenaa, 2022-02-16:
> ### Discussed in #1514
> Originally posted by **Denethor85** January 28, 2022 in the new version 2.6.0 there is no eeprom reset button, so how can I reset each setting to default? I tried reinstalling the firmware, wiping the SD card, but none of this seems to work ....
> 
> Thanks
> 
> Essentially, we need to brink back the reset to factory option, and then have some way to prevent a import of EEPROM on the reset. Perhaps an empty file that indicates clean slate is wanted, and is then deleted?

So the current behaviour is that the radio.yaml is generated from EEPROM when radio data is restored, correct?

raphaelcoeffic, 2022-02-17:
Without checking, I believe empty files would effect exactly this.

gagarinlg, 2022-02-17:
> Without checking, I believe empty files would effect exactly this.

Is there a difference between an empty file and a partially filled file? With an incomplete file, I loose all the hardware settings regarding switch tyes etc.

stephenaa, 2022-02-23:
> 

It seems to me as if reverting to EEPROM settings are done in sdcard_yaml.cpp:loadRadioSettings() if no radio.yml exists, so maybe the block of code governed by the if at line 196 should include a check for the case that radio.yml exists, but has size==0. If radio_new.yml exists reading that should be attempted. "Factory reset" then would have to delete both files and call loadRadioSettings().

gagarinlg, 2023-03-18:
2.9 does not support binary conversions anymore. This issue has been resolved by waiting long enough.

closed 2023-03-18 by gagarinlg