EdgeTX/edgetx #1528: 'cmake warning' Linux building

nostromo123, 2022-01-30:
Have two building environments on my computer. Besides Ubuntu 20.04 also Debian Bullseye.
Bulding 2.6 release from sources I see in both this warning.

It seems to be only a minor issue, this warning about empty variables, but could become serious with further cmake versions. See different egdeTX version under Debian.

Please see attaches pdf.

br KH

[cmake warning.pdf](https://github.com/EdgeTX/edgetx/files/7965922/cmake.warning.pdf)

pfeerick, 2022-01-31:
In both instances older versions of CMake are being used - we are using 3.21 from the kitware repos

This is the equivalent output via the docker build environment:

```
root@027fba8d4796:/src/build-X10# cmake DPCBREV=EXPRESS -DHELI=NO -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Release ../
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:19 (cmake_policy):
  The OLD behavior for policy CMP0023 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Git: /usr/bin/git (found version "2.33.0") 
-- EdgeTX 2.7.0 @ a052ba63
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- Python found, version: 3.6.9
-- Qt Version: 5.12.9
-- Foxlib found at /usr/lib/x86_64-linux-gnu/libFOX-1.6.so
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found SDL: /usr/lib/x86_64-linux-gnu/libSDL.so;-lpthread (found version "1.2.15") 
-- SDL Lib: /usr/lib/x86_64-linux-gnu/libSDL.so;-lpthread; Headers: /usr/include/SDL; Version: 1.2.15
-- Googletest will be downloaded
-- Adding support for PXX1 as internal module
Storage conversion: 219
TARGET simu/libsimulator: cpp compiler /usr/bin/c++ v7
-- Added optional gtests target
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
TARGET bootloader: cpp compiler arm-none-eabi-g++ v10.2.1
TARGET firmware: cpp compiler arm-none-eabi-g++ v10.2.1
-- Simulators library search path: ../lib/companion27
-- Performing Test FLAG_WEXTRA
-- Performing Test FLAG_WEXTRA - Success
TARGET companion27: cpp compiler /usr/bin/c++ v7
-- Added optional gtests-companion target
CMake Deprecation Warning at companion/src/CMakeLists.txt:354 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- install /src/build-X10 to /usr/local/bin
-- linuxdeployqt will be downloaded
squashfs-root/.DirIcon
squashfs-root/AppRun
squashfs-root/linuxdeployqt.desktop
squashfs-root/linuxdeployqt.svg
squashfs-root/usr
squashfs-root/usr/bin
squashfs-root/usr/bin/appimagetool
squashfs-root/usr/bin/desktop-file-validate
squashfs-root/usr/bin/linuxdeployqt
squashfs-root/usr/bin/patchelf
squashfs-root/usr/bin/qt.conf
squashfs-root/usr/bin/zsyncmake
squashfs-root/usr/lib
squashfs-root/usr/lib/appimagekit
squashfs-root/usr/lib/appimagekit/mksquashfs
squashfs-root/usr/lib/libQt5Core.so.5
squashfs-root/usr/lib/libicudata.so.55
squashfs-root/usr/lib/libicui18n.so.55
squashfs-root/usr/lib/libicuuc.so.55
squashfs-root/usr/lib/libpcre.so.3
squashfs-root/usr/share
squashfs-root/usr/share/doc
squashfs-root/usr/share/doc/libicu55
squashfs-root/usr/share/doc/libicu55/copyright
squashfs-root/usr/share/doc/libpcre3
squashfs-root/usr/share/doc/libpcre3/copyright
squashfs-root/usr/share/doc/qt59base
squashfs-root/usr/share/doc/qt59base/copyright
squashfs-root/usr/translations
-- Configuring done
CMake Warning (dev) at radio/src/tests/CMakeLists.txt:55 (add_executable):
  Policy CMP0115 is not set: Source file extensions must be explicit.  Run
  "cmake --help-policy CMP0115" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  File:

    /src/radio/src/tests/location.h.in
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /src/build-X10
```

nostromo123, 2022-01-31:
Thank you for answering.

I'll shift my Debian building environment from Bullseye to Bookworm (Testing) which ships with a cmake 3.2.2.1-1+b1

One add. question, I am building with the 2.6 release sources, but the generated *.bin file reports as 2.7! in the protocol and on the radio? That puzzles me a bit.

br KH

pfeerick, 2022-01-31:
Sounds like you're building the `main` branch, not the 2.6 branch ;)

closed 2022-01-31 by nostromo123