EdgeTX/edgetx #1529: background color depth

atazx, 2022-01-30:
**Describe the bug**
background image is not displayed correctly, maybe too low color depth? with opentx it was displayed correctly. the issue can also be viewed by taking a screenshot

**To Reproduce**
set a custom background image

**Expected behavior**
the image should be displayed correctly

**Screenshots**
[image]
[image]

**General information**
 - EdgeTX version 2.6
 - Hardware radiomaster tx16s

atazx, 2022-04-26:
still present also in version 2.7.
How can I solve?

rotorman, 2022-04-26:
The radio display is hooked up via RGB565, meaning 5 bits for red, 6 for green and 5 for blue, and thus the color output is somewhat limited, when compared to a normal PC with at least 8-bits per color.

See Sec. 2.7 of the TX16S schematic: https://www.rcgroups.com/forums/showthread.php?3869543-Blog-17-RadioMaster-TX16S-schematic-diagram

atazx, 2022-04-26:
Ok, but the colors on opentx are a lot better...
i just flashed opentx and took a picture.
please see the attached picture:

OPENTX
[image]

EDGETX
[image]

atazx, 2022-05-05:
As you can see from the two photos the ELRS logo is displayed differently, with lower quality on edgetx. The problem is therefore not limited only to the background but to all the images.
Should I change the title? or open another issue?
Am I the only one with this problem? how could i fix it? what other tests can i do?

pfeerick, 2022-05-05:
You can change the title as it applies across the board to PNG image handling. It has been mentioned before, and I think it was worked around by saving the PNG differently. AFAIK it is a limitation of the image processing code that OTX switched to for 2.4, which may or may no be overcome in the transistion to LVGL that is in progress.

closed 2023-09-05 by raphaelcoeffic