EdgeTX/edgetx #1531: nirvana NV14 flysky brick

yhomega, 2022-01-31:


eshifri, 2022-02-01:
I would say that flashing from the SD card is a very reliable way to do it.

pfeerick, 2022-02-01:
As long as you are flashing the NV14 firmware, I'm not sure what is going on here... i.e. double check after you select the firmware via the bootloader menu that it says "radio: nv14" at the bottom of the screen. However, I have just spotted an unrelated bug in relation to trimhat navigation in the 2.6.0 bootloader :-/

stefan73, 2022-02-01:
Additional thought: Some older FWs for the Nirvana hat partially unreliable flashing SW in the bootloader. If flashed via SD afaik the bootloader is not updated. I would recommend to update the radio to EdgeTX via DFU once.

yhomega, 2022-02-01:
Thanks for the answer, so stefan73 you recommend to flash using "STM32CubeProgrammer " or "opentx companion"? Using the buddy edgetx it's also DFU!

Pfeerick and Eshifri : When using the SD card version, I can't get to bootloader. It "bricks" just before that.

actually I just did some more testing, and I can get to bootloader with 2.5 and 2.6 rc1
With 2.6 rc2, rc3, nightly and 2.6 santa I get black screen, no bootloader, and I have to remove battery in order to boot the radio again.

eshifri, 2022-02-01:
If you can flash the 2.6RC1 FW through DFU you will get the working bootloader.
Now you can flash any other version from SD card - it will not overwrite the bootloader. (And your bootloader will be "good enough" :-) )
Now you can experiment with other FW versions.
As pfeerick has suggested the first thing to do is to check that you are flashing the NV14 version.

yhomega, 2022-02-01:
Eshifri, I just flash RC1 with edgetx buddy through DFU, It worked fined
Then I flash 2.6 from SD with RC1 bootloader, it bricks the nirvana.

stefan73, 2022-02-01:
Just retried. Flashed the old FW and then back to EdgeTX. Works.
Are you using old SD card contents? - Try with a fresh setup SD from here: https://github.com/EdgeTX/edgetx-sdcard/releases

yhomega, 2022-02-01:
yes I did format my sd and used the edgetx sdcard thing

I also tried rc1 with edgetx buddy, then 2.6 with edgetx buddy, brick also

gagarinlg, 2022-02-05:
Can you please send a picture of the mainboardand zip of your sdcard? I have two different hardware versions here and both work with the 2.6 release.

yhomega, 2022-02-05:
https://photos.app.goo.gl/SEGFWQTMFMP37r86A photos I did when I received the nirvana few months ago. Some sort of Power mod was done from factory. For the sdcard I format it before installing edgetx. Can't provide a zip right now

yhomega, 2022-02-05:
https://drive.google.com/file/d/1annH4WoD3M2fWzGG1BBMeYaqvx0MlNoa/view?usp=sharing here is the zip of the sdcard

gagarinlg, 2022-02-05:
Thank you, I will take a look, if there is anything obvious

yhomega, 2022-02-05:
The zip is from sd 2.5 but I tried with sd 2.6 and same issue

yhomega, 2022-02-05:
It's definitly something done on edgetx after rc1. Because 2.5 and 2.6 rc1 work fine.

gagarinlg, 2022-02-05:
> It's definitly something done on edgetx after rc1. Because 2.5 and 2.6 rc1 work fine.

I tried tried your sd card content with 2.6 on both NV14 I have, and both work.
Which country do you llive in?

yhomega, 2022-02-05:
france, NV14 was ordered on aliexpress

gagarinlg, 2022-02-05:
> france, NV14 was ordered on aliexpress

I do not know of a developer living in france, but some are living in Germany. If you are close enought to the broder, maybe something can be arranged, if nobody else is able to reproduce the problem.

yhomega, 2022-02-05:
I will have to ship it to germany or an other country... if I can avoid this. Would you like a video? Or I do some sort of debugging maybe? Meanwhile you recommend me to stick to 2.5 or 2.6 rc1? You saw the pictures of my hardware. Yours is similar with power mod from factory? It could be the issue maybe?!

gagarinlg, 2022-02-05:
> I will have to ship it to germany or an other country... if I can avoid this. Would you like a video? Or I do some sort of debugging maybe? Meanwhile you recommend me to stick to 2.5 or 2.6 rc1? You [...]

Do you have access to a SWD debugger, ST-Link or something similar? A video probably wouldn't help much.
Both of my NV14s look different. One has a much newer main PCB, the other one has the same as you have PCB, but the wiring looks different. 
I'll take a look at a diff between rc1 and 2.6, maybe I can spot something.

yhomega, 2022-02-05:
I must have a ST link somewhere, just can't find it right now.
How hard is it to use it on the nirvana? 
CHeers

gagarinlg, 2022-02-05:
You have to find the pads for SWCLK, SWDIO, NRST, MCU_3_3V and GND and solder a cable on them. Attached as a picture of one of my NV14. The other one is closed currently.
[image]
.

gagarinlg, 2022-02-05:
In the diff between RC1 and 2.6 release I found nothing obvious regarding NV14, that could break stuff like that.

rotorman, 2022-02-05:
One hint about ST-Link. If you do not have the original ST-Link (looks like this: https://www.st.com/bin/ecommerce/api/image.PF251168.en.feature-description-include-personalized-no-cpn-large.jpg ) and have a clone, you might want to omit connecting the +3.3V line. The reason is that orignal ST-Link uses this to set the voltage levels, it is a reference input for the ST-Link. Numerous clones output +3.3V on that line. It would be then multiple voltage regulators operating on the same line, which is not good.

yhomega, 2022-02-05:
thanks, I have a race tomorrow, I will try to do it on tuesday/wednesday. What should I do after connection?

yhomega, 2022-02-16:
Any recommendation after the stlink is connected? I just try to update again or something else? Cheers

rotorman, 2022-02-16:
First test would be if STM32CubeProgrammer detects the microcontroller at all. If so, then the good news is that the microcontroller is still alive. You can next erase the chip and flash EdgeTX NV14 binary with STM32CubeProgrammer via SWD.

yhomega, 2022-03-20:
ok I did it but exactly the same problem as flashing with edgetx buddy or the sdcar
[image]
d

rotorman, 2022-03-20:
The STM32CubeProgrammer is happy in your case - verify successful. Can you enter EdgeTX bootloader at least with your NV14?

yhomega, 2022-03-20:
yes like previously

yhomega, 2022-03-20:
Im about to remove the stlink wires inside and just give up. Anythink else I can do?
I think I will get a new nirvana. gagarinlg does your nV14 with v2.2 motherboard needs some sort of powermod or any other mod to work properly? (Removing pull-up on S.PORT line?)

thanks

gagarinlg, 2022-03-20:
> Im about to remove the stlink wires inside and just give up. Anythink else I can do? I think I will get a new nirvana. gagarinlg does your nV14 with v2.2 motherboard needs some sort of powermod or any other mod to work properly? (Removing pull-up on S.PORT line?)
> 
> thanks

I never really did anything with it except debugging. I connected the MPM externaly and short after the voltage regulator died. Ever since I only used it for debugging and applying 3.3V from the inside.

rotorman, 2022-03-20:
@yhomega as you can enter bootloader, the display and the connection works, so the issue is just that the main firmware for some unknown reason does not display anything on your display. Rename on SD card \RADIO to \RADIO_bck and \MODELS to \MODELS_bck and remove any custom splash or themes, if you have installed any. Does this help?

As you have ST-Link attached, you could try to let the code run in in-circuit-debug mode by stepping through the initialization and see where on earth it hangs in your case.

gagarinlg, 2022-03-20:
If you are able to debug, please try to check which display controller you have by stepping the the coder in targets/nv4/lcd_driver.cpp

yhomega, 2022-03-20:
thanks for the inputs but I have no idea how to debug and I have already removed st link wires. I ordered an other nv14 and will send the faulty one back

rotorman, 2022-03-20:
Hmm... but you said it worked with EdgeTX 2.5 w/o issues, so it might not be a hw problem?

yhomega, 2022-03-20:
so what's the problem? everybody is able to flash 2.6 Im the only one having issue

yhomega, 2022-03-20:
its not just a display issue. the nv14 brick once I flash 2.6
the screen very slightly turn brightness on, but on/off don't work, only removing the battery work

yhomega, 2022-03-20:
2.6 rc1 did work, but not above

eshifri, 2022-03-20:
Did we change the display driver between 2.6RC1 and RC2?

rotorman, 2022-03-20:
This is the last change to NV14 display driver on Dec. 26th, 2021: https://github.com/EdgeTX/edgetx/commit/a1e03ed7b5745609134f4425e68fdbbf573233c5

gagarinlg, 2022-03-20:
There was a change to fix one display type.
Maybe it would help to put change line 1344
  } else if (LCD_ST7796S_ReadID() == LCD_ST7796S_ID) {
  to
  } else {
and remove the unknown lcd controller block, but why does the bootloader work? it uses the same code

gagarinlg, 2022-03-20:
@yhomega can you please try the attached build? You have to unpack it before flashing.

[firmware.zip](https://github.com/EdgeTX/edgetx/files/8311989/firmware.zip)

yhomega, 2022-03-20:
yes it works thanks. I should keep it?

gagarinlg, 2022-03-20:
If you like. This is the current development state plus the fix
It will be fixed in 2.7

closed 2022-03-21 by pfeerick

gagarinlg, 2022-03-21:
@yhomega it is in the nightlies now.
Thx @pfeerick

yhomega, 2022-03-21:
thank you very much I will enjoy expresslrs/edgetx fast communication :)
regarding sdcard, I can use the content from 2.5?
cheers

gagarinlg, 2022-03-21:
Themes have a different format. The radio and model settings will be converted to Yaml.

yhomega, 2022-03-21:
For themes it's not an issue I don't use them.
For radio and model setting, conversion is automatic or I have something to do?

gagarinlg, 2022-03-21:
Should be automatic. If you need any help, join the discord server at https://discord.gg/wF9wUKnZ6H

PR #1714 (2022-03-21): fix #1531, not wokring screen on NV14 with ST7796S display controller
Fixes #1531

Summary of changes: assume ST7796S display controller on NV14, when no other is detected