EdgeTX/edgetx #1532: X9D ACCST No possibility to change the internal module?

bbc25185, 2022-01-31:
Good morning.
With firmware version 2.6, I noticed the addition of being able to change the default internal module.
So I flashed edgetx version 2.6 stable on my X9D (first ACCST version) and I realized that the entry to change the internal module is NOT present.
Is there some reason that escapes me why that voice is not present ??? (obviously I haven't unsoldered the original module yet, could this be the reason?)
Will it be added in the future ??? I would like to be able to replace the original XJT module with a MultiModule that I already own and free the module bay for an ELRS module!!!

bbc25185, 2022-03-23:
In some radios the radio module (the X9D plus version certainly) are a single electronic backboard and not possible to replace the internal radio module (not easily).
on mine (one of the first versions), it is literally the electronic board of an XJT module soldered onto the backboard ... therefore "easily" replaceable with another radio module. (i dont have a photo, but like this: https://www.baronerosso.it/forum/attachments/radiocomandi/377274d1578763564-frsky-taranis-x9d-problema-antenna-img_20200111_180105.jpg)

raphaelcoeffic, 2022-03-23:
That looks funny! Never seen that before! Small issue however: the internal connection is not hardware serial. So your best bet is on the S.PORT pin, which is the same use on the external bay by ELRS modules.