EdgeTX/edgetx #1541: ETX 2.6 Companion + Zorro: Spurious message 'File board ... does not match current profile board'

RC-SOAR, 2022-02-01:
When transferring models from Zorro transmitter into ETX Companion 2.6 I get an error message, I think it's spurious.

To reproduce:

1. Connect to Companion via USB
2. Menu 'Read Models and Settings from Radio'
3. Then I get this message

[image]

After dismissing the message, the models are transferred as expected.

Screenshot of Companion settings:
[image]

elecpower, 2022-02-01:
That is a bug. Will need to check whether the issue is on radio or Companion side

pfeerick, 2022-02-01:
I can't reproduce this - when importing from zorro it is fine. Since it says "file board", can you share your `RADIO/radio.yml` file?

RC-SOAR, 2022-02-02:
Sure, radio.yml file attached. Thanks for looking at this.

[radio - yml.txt](https://github.com/EdgeTX/edgetx/files/7982679/radio.-.yml.txt)

elecpower, 2022-02-02:
The radio.yml file is missing the board entry and for historical reasons it assumes TX16S when it should say "unknown"

pfeerick, 2022-02-02:
Ok, so the error is coming up because the `board:` entry is missing completely... i.e. usually within the first three lines (it can move around a bit) you'll have `board: zorro` which is what Companion is looking for to work out which radio config it's reading... so given this seems to be stemming from #1542, there are probably two bugs present... 
1) when `board` entry is missing, Companion assumes TX16s rather than saying... Unknown?
2) the board entry isn't being written when trying to convert by switching the profile

elecpower, 2022-02-07:
2. The code always writes the current profile firmware board to the radio.yml file.

PR #1572 (2022-02-08): Companion improve read radio settings messages
Fixes #1541 and mentioned in part in 1413

Summary of changes:
Improve messages and remove reference to conversion as there is no indication of how successful the conversion was giving the user the impression all is ok when it may not.
OTX file format conversion is very detailed.
closed 2022-02-08 by pfeerick