EdgeTX/edgetx #1542: Corruption when migrating TX16S setup to Zorro

RC-SOAR, 2022-02-01:
After migrating a setup from the TX16S to Zorro via Companion, Zorro appears to do an internal conversion which corrupts REPLACE and MULT mixes.

1. Start ETX Companion, connect to Zorro
2. Using TX16S profile, open an TX16S .OTX file containing a setup called, say 'target', which has REPLACE and MULT mixes.
3. Change to Zorro profile. The mixers in 'target' translate fine (except see #1541)

At this point, everything looks good, so time to transfer 'target' to the Zorro. So

4. 'Read models and settings', add 'target' to model list, then 'Write models and settings'. So now Zorro has model 'target'
5. Read models and settings from Zorro. 'target' mixers look fine, so it should be fine on the transmitter
6. Disconnect USB

At this point, I start the transmitter, and find that all the REPLACE and MULT mixes in 'target' have been changed to ADD. 

All the above has been using .ETX files, as I cannot save to .yaml with either the Zorro or TX16S profiles (the save as yaml option is present, but selecting it does not actually write a file).

pfeerick, 2022-02-02:
Can you add a copy of the offending model.yml from the TX16s and the resulting Zorro model.yml, so we can figure out if it's companion or firmware that is responsible...

RC-SOAR, 2022-02-02:
Attached is the problematic yml file, directly off the Zorro.

On the TX16S, I'm running OpenTX on the TX16S so can't offer a yaml file for the original model.  The .OTX file containing the original model was generated in OpenTX Companion 2.3.14 via 'read models and settings'. I then created a new document, still in Companion, with just that one model, and save that as an OTX. I then opened that file in ETX Companion to begin the migration process.
[model05_buggy_off_zorro_yml.txt](https://github.com/EdgeTX/edgetx/files/7982885/model05_buggy_off_zorro_yml.txt)

pfeerick, 2022-02-02:
Oh, derp, yeah, ok, that makes sense now... so it's probably the "conversion by profile switching" in OTX that's not working properly. 

I just mocked up a simple OTX with a couple of multi and replace mixes, and it seemed to convert fine... so I'm wondering if it is also related to the radio read and write... curiously... it also did add the zorro board parameter. 

What happens if you open your tx16s otx, switch to zorro, and then save it to etx, and then re-open the etx... does it still look fine then? I also tried just going from otx to etx, and that seemed fine also... 

(just remove .zip file extensions)
[tx16s-add-multi-replace-test.otx.zip](https://github.com/EdgeTX/edgetx/files/7983260/tx16s-add-multi-replace-test.otx.zip)
[tx16s-add-multi-replace-test.etx.zip](https://github.com/EdgeTX/edgetx/files/7983259/tx16s-add-multi-replace-test.etx.zip)
[zorro-add-multi-replace-test_converted.etx.zip](https://github.com/EdgeTX/edgetx/files/7983261/zorro-add-multi-replace-test_converted.etx.zip)

RC-SOAR, 2022-02-02:
Thanks. Will try when I get home in the evening UK time.

RC-SOAR, 2022-02-03:
I attach the .OTX file which started the trouble (remove .zip extension). Open it in Companion 2.6 and it should produce the error in #1541 , starting the chain of events leading to the corruption.

My feeling is that the error is occurring in the transmitter. This is because, despite the error message on initial loading into Companion, (1) the mixers are fine _before_ transferring to the tx (2) the mixers are fine _after_ being read back into Companion immediately after transfer. (2) However, after disconnecting the USB and booting the transmitter, the REPL and MULT mixers have been altered. (3) If the setup is now transferred back into Companion, the mixers are shown as altered in Companion.
[precision_adp_TS16S.otx.zip](https://github.com/EdgeTX/edgetx/files/7998969/precision_adp_TS16S.otx.zip)

gagarinlg, 2023-03-18:
Is this still a problem with 2.8? (I do not have a Zorro)

closed 2023-05-26 by gagarinlg