EdgeTX/edgetx #1544: Ghost Raw 12 Bits not working when box is check.

rcquad808, 2022-02-02:
Ghost "Raw 12 bits" not working on quad. 
Gost Updater v1.7
Version v1.0.5.2 BETA 12-bit on Ghost Hybrid V2 Uno
Using EdgeTX edgetx-firmware-merge #fdc1945 (2022-02-01T03:56:26Z)
Betaflight 4.3.0 RC2 firmware
Radiomaster TX16S Max

The quad is bound and my radio see it. When I turn on the 12 bit on EdgeTX the link is cut, but when I uncheck the 12 bit box the link is connected and everything is working fine.

pfeerick, 2022-02-02:
I don't believe 12bit support is present in Betaflight 4.3 RC2? - it seems like the PR to add that has been scheduled for Betaflight 4.4? 

@daleckystepan Is that right?

daleckystepan, 2022-02-02:
@pfeerick You are right.
@rcquad808 It has not been merged yet to BF 4.3 (RC phase, feature freeze). You need to wait for BF 4.4 or you can compile your own build.

closed 2022-02-03 by pfeerick