EdgeTX/edgetx #1552: Frsky XLite Error updating firmware

asudhir1947, 2022-02-03:
Description:

 Attempted to update the firmware of the internal module of Frsky Xlite from 2.1 Fport version to pre-2.1 non-fport version to work with XM+ receivers. 

I get an error saying 'Error updating firmware' with an option to 'Exit'. 

Version: EdgeTX version 2.6 
Radio: Frsky XLite 

Did not have this issue with EdgeTX 2.5

Kikdiweg, 2022-04-27:
Same problem here, module is not working too

eBulldog, 2022-05-03:
I have the same issue, EdgeTX v2.7.0 is unable to flash the internal iXJT RF module of the Frsky X-Lite.

- Firmware is stored on SD card
- Select file and option to "Flash Int. Module"
- "Device reset..." is displayed and within 5 seconds beeps
- Following beep you get error "FW update error" "Device not responding"
- Reverting to OpenTX and flashing process works as expected
- Video of process and failure available on request.

asudhir1947, 2022-05-03:
Yeah, surprised devs did not pick up, another issue I have seen is when I attach a betafpv tx module, it won’t even let change frequency all the way upto 500, it stops at 250 while on 2.5 I can easily set 500hz, that’s why I am still on 2.5, most stable I feel

pfeerick, 2022-05-04:
>  it won’t even let change frequency all the way upto 500, it stops at 250 while on 2.5 I can easily set 500hz

Are you increasing the serial baud rate up to above 400k for the external module? As I think the 250hz step is the highest you can go without increasing the serial baud. Anyway, if you have a different issue, it's best to open that it's thread. 

> Yeah, surprised devs did not pick up,

That would be because the "devs" don't have every hardware combination, and the "testers" obviously didn't test every combination. Those testers being as many community members as we can get to test the firmware during the release candidate (or before that, during specific pull requests) phase for hardware that we don't personally have in addition to tests on our own radios.

This has been the only report I've seen about 2.6 having issues (as well as it being unfortunately specific to the Frsky Xlite) with firmware updating. It seems to have been more a problem in 2.7 with external modules after the serial driver re-write (https://github.com/EdgeTX/edgetx/issues/1764) - there is most likely just some small difference in the timing - or even the power reset cycle. Give it time to be resolved... it's just a matter of the right people and hardware being able to work on the problem.

asudhir1947, 2022-05-04:
I can't change the baud rate above 400k as then I get a telemetry lost warning when using my betafpv ELRS module, from what I understand it is a hardware restriction, i don't actually have a telemetry issue, but it sets of this warning, a lot of folks have had this warning on the elrs facebook group and the only solution has been to set it to 400k baud rate. 

No offense to any of the devs/testers, I read the release notes and the hard work being put in by folks whose primary job is not developing this software and do this as a hobby, I am all for them, didn't intend to discredit any hard working devs or testers who are on this project, but if you need one more tester who has an xlite, I am ready to test.

pfeerick, 2022-05-05:
> No offense to any of the devs/testers, I read the release notes and the hard work being put in by folks whose primary job is not developing this software and do this as a hobby, I am all for them, [...]

No offense was/is taken. I do understand it can be quite frustrating... I may have some bug bears of my own over the woeful state of support for the 9X/9XR-PRO on OpenTX, which features were added for, but since they don't test anymore at all, Plus the radios are unusable with 2.2 and 2.3 of OTX (as in my book the radio firmware being prone to crashing isn't a feature). And there is then the whole touchscreen support for my TX16S in 2.4 :grin: Anyway, that is old news. 

Thanks for splitting to a separate issue, hopefully we can get to the bottom of that one sooner rather than later :) Unfortunately that is probably in part a hardware limitation, but will focus on it over there. #1935

And yes, more people who can test the better, so I'll keep you guys in mind. I'm slowly trying to get a list together of people who are willing to be called upon as testers for different hardware so we can get better coverage.

Nerenye, 2022-06-25:
same flashing problem here on xlite S with 2.7.1
if you need another tester. hit me up.

raphaelcoeffic, 2022-10-16:
@Nerenye I just tested this on a X9-lite, which is basically the same when it comes to the internal module connection. Can you please re-test with the current 2.8.0-rc3?

Nerenye, 2022-10-17:
@raphaelcoeffic it´s fixed. Tested with x-lite S on 2.8.0-rc3

raphaelcoeffic, 2022-10-18:
@Nerenye now that is really strange.... Does it actually work with OpenTx? We are using basically the same code here. Otherwise I might have to get hold of a Xlite-S to test it myself...

Nerenye, 2022-10-18:
On latest opentx I doesn't work. Tested that yesterday. Last working version I found was edge TX 2.5 how do I turn it off? Under m/modules it shows off all the time?

pfeerick, 2022-10-18:
Check what the SYS/RADIO -> HARDWARE -> Internal RF setting is... it could be set to OFF also, meaning the radio thinks it has nothing to turn on.

Nerenye, 2022-10-18:
I've the remote not at hand but yesterday that freaked me out thinking I bricked the radio.
It shows the internal module always as off, despite it working correctly and binding.

pfeerick, 2022-10-18:
Yes, the version page is different - that is just information - I meant the HARDWARE PAGE ;) 

[image]

However, it depends on what version of firmware you are running... I think 2.6 was the version that introduced that being configurable option? But, if you have it off in the model settings... like so...

[image]

... then there is something iffy going on still. The fact that OTX isn't working for you suggests there is a common bug lurking somewhere... which is why it's interesting that ETX 2.5 _is_ working for you. At least that gives us a frame of reference as to a working version to work forward from.

Nerenye, 2022-10-18:
I'm at the moment on 2.8 rc3.

I'll check that later. Could be that I missed the hardware setting because I had my sim model loaded which obviously has no rf setting. Switched later to a model with rc and checked the link which was working.
Its a bit strange that system settings would change on a per model basis. Was that supposed to be?

I tried to flash with latest otx 3.15? Was it I think.
But that failed  several times aswell.
I then checked etx 2.5 because of the Thread opener and that worked out of the box.

Anything I can test for you? Any versions or so. I would send the radio to you but unfortunately it's the one of my daughter and she would not be happy .

pfeerick, 2022-10-18:
Most likely just haven't found the setting yet ;) System settings won't change on a model basis. The system level setting is there to support modding/radios with different internal RF modules as standard... i.e. if you change the internal ISRM RF module to a CRSF one,  you would change the setting there,  and then should get CRSF and OFF as options for the internal RF when in model settings. 

OpenTX 2.3.15 perhaps? That's the current version of that.

No need, Raphael is organising getting his hands on one ;)

Nerenye, 2022-10-18:
2.3.15 right that's the one.

As soon as you have a commit to checkout and build hit me up.

Thanks for your effort!

raphaelcoeffic, 2022-10-18:
@Nerenye PR #2596 should fix the issue. Would you mind re-trying *again*?

raphaelcoeffic, 2022-10-18:
By the way, I believe this might affect *any* radio with a `iXJT` (built-in X12S & X10 first gen)  or `ISRM` module inside (any ACCESS radio or update ISRM).

Nerenye, 2022-10-18:
Just sold my x12s and got myself a tandem x20
 so I'm only be able to check with the xlite s

Nerenye, 2022-10-18:
@raphaelcoeffic it´s fixed! Testet with your PR on X-lite S 
Sys->Hardware shows the internal module as isrm 
Flashing works as it should.

Thanks a lot for your effort!

PR #2596 (2022-10-19): fix(isrm): flashing internal ISRM module
Fixes #1552

Summary of changes:
- properly handle flashing of internal module depending internal module set in radio settings
- deactivate UART on internal serial module as the module won't reset properly otherwise
- fix a minor issue with the authentication counter not properly reset for ISRM upgrade modules
commit 20671405 (2022-10-19): fix(isrm): flashing internal ISRM module (#2596)

Fixes #1552
commit 5ebfa5c2 (2022-10-31): fix(isrm): flashing internal ISRM module (#2596)

Fixes #1552
closed 2022-10-19 by pfeerick