EdgeTX/edgetx #1553: Jumper T18 Pro running EdgeTX v2.6.0 causing flight glitching

JoeSa3rd, 2022-02-03:
**Describe the bug**
I recently upgraded my Jumper T18 Pro from EdgeTX v2.5.0 to v2.6.0 and now I'm experiencing glitching in flight. When I rollback to FW v2.5.0 the glitching issue goes away. Also I've never had this issue when running v2.5.0 and I've been using v2.5.0 since the week is came out. As another test I flashed my Frsky Taranis X9D Plus with EdgeTX v2.6.0 and flew the same quad with zero glitching issues.

**To Reproduce**
Flash Jumper T18 Pro with v2.6.0

**Expected behavior**
No glitching

**Screenshots**
Video showing the glitching issue -> https://www.dropbox.com/s/4yvdno71e06e3l7/EdgeTX%20v2.6.0%20Glitching.MP4?dl=0

**General information**
Jumper T18 Pro running EdgeTX FW v2.6.0, TBS Crossfire TX FW v6.13, 250mw, Dynamic Power Enabled
Quad running Betaflight v4.1.2, TBS Nano RX FW v6.13, CRFS

pfeerick, 2022-02-04:
What triggered the OSD elements to be turned on just after that first glitch happened at 0:12? I'm assuming that is just coincidence? Have you double checked the MPM frequency tuning is still ok (although I'm not seeing RX_LOSS message there so I also doubt that is it. See what happens if you turn off (or on, if it's off) the ADC filter on the RADIO/Hardware tab... ideally for a quad it should be off, as you want the flight controller to handle any jitter in the inputs...

JoeSa3rd, 2022-02-04:
I believe that's the TBS fusion OSD reporting what it believes is a degradation of LQ at some threshold. To be honest I've never seen that before and was just as puzzled. I just verified a second time that ADC filter is off but I will toggle it on and retest to see if anything changes.

closed 2023-09-05 by raphaelcoeffic