EdgeTX/edgetx #1556: Mixer output is negative instead of positive

RC-SOAR, 2022-02-04:
The setup has two channels, with the same aileron input. One channel has a negative weight. The other has a positive weight.

_Expected_: mixer outputs should be of opposite signs
_Actual_: mixer outputs are the same sign.

To see this, load attached .ETX file  (remove .zip extension) into ETX Companion 2.6 and run the sim.

[f3f_A2FBug.etx.zip](https://github.com/EdgeTX/edgetx/files/8005696/f3f_A2FBug.etx.zip)

Some background: the attached setup originated from OpenTX/TX16S. The .OTX file was was opened in ETX companion, and converted to .ETX format before being attached.

pfeerick, 2022-02-05:
This may be related to/have been fixed in #1516, if so will be in 2.6.1 Will check the ETX later to confirm.

pfeerick, 2022-02-05:
Ok, yes it is. If you check the GVAR that simulator is using for the mixer weight, you'll see it's using the wrong one for the -GV, hence why it is behaving that way. The radio firmware will similarly mangle it, so you will need the 2.6.1 firmware to fix this.

https://user-images.githubusercontent.com/5500713/152636969-a049ffaa-ed6b-44a1-ba2a-e1d073f32cc3.mp4

RC-SOAR, 2022-02-05:
Great, thanks for fixing.

pfeerick, 2022-02-05:
Credit goes to Jesper @jfrickmann as he did the PR fixing it last week ;) I'll close this as it is already merged and already scheduled for 2.6.1

closed 2022-02-05 by pfeerick