EdgeTX/edgetx #1557: Companion 2.6 doesn't read/write the parameters of the telemetry screens on X7/X9D+

nmare99, 2022-02-04:
Companion 2.6 doesn't read/write the parameters of the telemetry screens from/to the transmitter.
This was revealed on taranis X7 and X9D+
Screenshots of the companion and the edgetx screen are attached

[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]

pfeerick, 2022-02-05:
Confirmed with T-Lite also - when reading Companion is mangling the telemetry screen data:

[image]

This zip from the radio when loaded in Companion is also demonstrating the failure. 
[radio.zip](https://github.com/EdgeTX/edgetx/files/8007724/radio.zip)

elecpower, 2022-02-06:
Issue also affects screen type Bars

PR #1562 (2022-02-08): Companion fix b&w custom telemetry screens yaml
Fixes #1557

Summary of changes:
Add union struct on read and write.
Alternative array iterations method on write.
closed 2022-02-08 by pfeerick