EdgeTX/edgetx #1558: Jumpet T-Pro: Impossible to setup telemetry screen

SergeyBokhantsev, 2022-02-05:
**Describe the bug**
Telemetry screen is not working

**To Reproduce**
Precondition: ELRS/CRSF connection, telemetry sensors discovered and working.

In Companion:
Custom telemetry screen configured like following:
[image]
Then upload config to radio.
Result: empty telemetry screen on radio, "---" on the "Display" page config tab.

In Custom Telemetry Screen page on the Radio (the "Display" page tab):
When trying to customize the screen there is no possibility to select a source other than SW1:
[image]

**General information**
 - EdgeTX version 2.6.0
 - Jumper T-pro, BetaFPV ELRS Nano 900

pfeerick, 2022-02-05:
When you say you can't pick any source other than SW1, do you mean on the transmitter? As I can't reproduce that... I can pick any of the sources - inputs, swithes, sticks, trims, etc... Otherwise, this seems to be a duplicate #1557

SergeyBokhantsev, 2022-02-06:
@pfeerick yes, that's right. It behaves like the SW1 is in active state permanently. So whenever you try to set any another source - it just replaces it back to SW1 in a half of second.
Thanks for noticing me on #1557 looks like the writing telemetry screens to the radio is the separate unrelated issue.
So I will check the SW1 problem again to see if I could find additional details.

pfeerick, 2022-02-07:
Ah, ok, that is probably this issue then #1488, and I'm not seeing that behaviour on mine on my atm since I have them turned off atm and didn't realise telemetry screen also used also used autosource to pick values... 😆

closed 2023-09-05 by raphaelcoeffic