EdgeTX/edgetx #1563: Trainer Mode Master/Multi in X10s

elekSENDER, 2022-02-06:
I´ve understand there is a transfer issue for Trainer Settings between companion and radios.
But I noticed an issue in my Horus X10s which I think is not related to companion.

**Describe the bug**
I can select Trainer Mode in the X10s to Master/Multi.
Master/Multi makes in the X10s afaik no sense due to hardware limitation.

**To Reproduce**
Select Master/Multi in Trainer Mode

**Expected behavior**
No possibility to select Master/Multi

**Screenshots**
[image]
[image]

**General information**
 - EdgeTX version (2.6.0 x10-338d4d2)
 - Hardware (Frsky Horus X10s)

ZS-Man, 2022-02-14:
Hi,
Multi trainer is possible on X10S express with external MPM module and minor HW mod on radio RF module.
If this option will be removed, leave the option for ETX manual compilation, please.  (-DALLOW_TRAINER_MULTI=YES   ?)
Thanks.

pfeerick, 2022-02-14:
It's more likely it will be left as it is if that is the case, or possibly an option added to the HW page so a custom build isn't needed.

ZS-Man, 2022-02-15:
There is info about this HW modification:

https://www.rcgroups.com/forums/showpost.php?p=44546553&postcount=17686

https://github.com/opentx/opentx/issues/7181

pfeerick, 2022-03-26:
Given that:
- when using an external MPM, Master/Multi is a valid option
- when using a radio that either as stock, or after a simple modification MPM, "Master/Multi" is valid

I propose the following:
1) that companion be a bit smarter - if a external MPM is configured, "Master/Multi" should be an option
2) that the firmware also be the same... i.e. it already knows what internal module is set, so the same can happen for external (if this is not the case already, that is)

elekSENDER, 2022-04-02:
Yes, makes absolutly sense, good idea.
So you have just one firmware adjusted by parameter.

commit be9cc540 (2022-09-14): fix: Don't allow select MULTI_TRAINER if no MULTI
Fixes #1563 by ensuring Multi Trainer opt is not shown if no MPM
is enabled, as well as if MPM suport is not even compiled in.
PR #2299 (2022-09-14): fix: Don't allow select MULTI_TRAINER if no MULTI
Fixes #1563 by ensuring Multi Trainer opt is not shown if no MPM
is enabled, as well as if MPM suport is not even compiled in.

It should only prevent `TRAINER_MODE_MULTI` being an selectable trainer mode if:
- `MULTIMODULE` support not enabled
- You don't have any modules enabled at all
- Neither of the enabled modules are MPM

It seems that having two MPMs enabled is a valid case, so I've left that case alone.

Hardware tested on TX16S and TX12-MK2, and working as expected. 

For me, this is both a bug (letting you select something that you c [... cut]
commit d54e6566 (2022-09-14): fix: Don't allow select MULTI_TRAINER if no MULTI (#2299)

* fix: Don't allow select MULTI_TRAINER if no MULTI
Fixes #1563 by ensuring Multi Trainer opt is not shown if no MPM
is enabled, as well as if MPM suport is not even compiled in.

* chore: Uppercase "OFF" or equiv for Trainer OFF
closed 2022-09-14 by pfeerick