EdgeTX/edgetx #1565: Radiomaster TX16S boot stuck on black screen

daleckystepan, 2022-02-06:
**Describe the bug**
Radiomaster TX16S stuck during the boot on black screen.
In some cases (don't know exactly) it will boot in a minute.

**To Reproduce**
Boot TX16S a few times in a row. No regular pattern.

**Expected behavior**
Boot to UI each time.

**Screenshots**
Video from another guy with the same issue
https://streamable.com/68l583

**General information**
 - EdgeTX version: 2.6.0 RCs, latest master, probably whathaver after 2.5
 - Hardware: Radiomaster TX16S, no modifications
 
**What I have tried**
- Discussed on Discord - more people have the same issue
- Flash different ETX version, compile myself, prepared build ... no difference
- Erase and create SD card content from scratch

pfeerick, 2022-02-06:
Conversation was around here: https://discord.com/channels/839849772864503828/839849772864503832/937346874220888084

Another user commented that they thought something similar started happening with 2.5, but was ok on 2.4. I believe also when it does start after that delay the radio settings are mucked up / back to defaults?

RipperGOT, 2022-02-06:
Try with different SDcard.
I had this problem a few months ago with 2 SD cards, one due to poor condition and the other due to low reading speed.

daleckystepan, 2022-02-06:
Today I have reformatted stock SD (FAT16) and create content from scratch. I will let you know. Than I can try different SD.

scooper1, 2022-02-12:
I have seen this black screen issue on startup a few times on a new eachine tx16s which is the same internals as radiomaster

Flierman1945, 2022-02-13:
Hi, my problems with no start up on the radiomaster started when I moved over to Edge.  not that Edge is the problem, but I was burning loads of cards and then suddenly this problem arose.Even when I bought the Eachine, it was the same.  Every one said I was building the image wrong, formatting wrong and trying their best to offer advice.  When ver 2.6 was released. I wrote new cards and the same old problem surfaced, until I checked the maker and every Samsung card I have will not start the Tx without my 40 second delay.  After that it is fine. Some times after restarting the problem will disappear till the next reboot.  But using Scandisk card seems , for me, the problem is fixed on the Rmster and Eachine Tx

scooper1, 2022-02-13:
A little morn diag   problem occurs if you  release the power button before the mode LEDs  start scanning.   This is on 2.6

pfeerick, 2022-02-14:
As in the timing seems important? i.e. I usually hold the button down until the speaker pops and LCD flashes, and release then, but the mode leds still does one more scan after that. Maybe we're running into a power issue again on some hardware, similar to with the MPM... i.e. trying to do much too quickly at boot, and cause a bit of a power brownout at just the wrong time.

fadius, 2022-02-21:
I have the same issue on Eachine TX16S. At first I thought it was happening because I was releasing the power button too early so I started waiting for EdgeTX screen to appear before releasing. It *may* have helped but the problem still occurs.
I'm using the stock 256MB HC SD card that came with the radio (no name).

Flierman1945, 2022-02-21:
Ahh, I have experienced this as well, a card that has worked and now won't start  in the Tx for nearly a minute.  All I did was to download the basic sd card content from the edge site.  Format your card again (after backing all the content up to a pc).  now drag over the files from the downloaded edge site and also your models and images etc and rewrite again to sd card.  It is a pain, but if you try some other cards like Scan-disk, they worked for me. 
 I think there is an underlying problem with the radiomaster and Eachine, it was mentioned the possible fault might be a brown out where at start up too much power is drawn by the processor and the system suffers a glitch.

raphaelcoeffic, 2022-02-21:
Someone is sending me one of these cards so I can test with it. We will see ;-)

GeorgiaBorn, 2022-05-05:
Having same issue with 2.7 on radiomaster tx16s. Around 45-60sec boot time. Using evo 32g sd card. When it boots up no issues everything works as it should. I just wait it out for now but would be nice if it was faster.

Flierman1945, 2022-05-06:
Hi, I think there is a fundamental problem with the 16Tx series.  I remember a dev suggested it might be the loading in the cpu at start up and the power supply drops a bit and there is a brownout. I found Scandisk 32 gig cards work best for me and you must hold your finger on the power button until there is a click form the speaker. Then it will power on every time.
I still have the problem if i don't follow this guide and it is the same on my Eachine 16  tx .
I too used the Samsung cards and they seem pone to this problem.  Hope this helps

daleckystepan, 2022-05-06:
I haven't experienced this issue for a months. No issues so far. What I did:
* Reformatted SD card from scratch - still using the stock one
* Updated regularly once a week to the latest nightly

pfeerick, 2022-05-06:
That was probably me. If it is brownout related, there is a patch in place that should prevent this as of 2.7. However, the common thread so far seems to be more the use of Samsung EVO cards, which I myself have used also... so it is either just a unforuntate coincidence or some weird compatabilty issue. I have to admit I think I'm using stock SD cards again and havn't had this for a long time now... and I also follow the "hold down the power button until speaker click / screen has flashed" methodology for what it's worth also...

GeorgiaBorn, 2022-05-10:
Update: changed sd card to a basic SanDisk 2gb works perfectly for now. Not sure its a speed issue or size issue. Question, has anyone ever used up 2gb?

rotorman, 2022-05-10:
I have been using 32 GByte SanDisk High and Max Endurance microSD cards, formatted with FAT32, without seeing a more than couple of second boot time.

GeorgiaBorn, 2022-05-10:
Was using a Samsung Evo 32gb, my boot times were up to 60sec at times.

pfeerick, 2022-08-19:
#2210, which will be in the next nightly build, will hopefully resolve this if anyone affected wishes to test and comment.

gagarinlg, 2023-03-18:
Is this still an issue?

scooper1, 2023-03-18:
this is still an issue with an ea tx16 if you don't keep power pressed till the screen starts but might not have #2210 installed yet

pfeerick, 2023-03-19:
#2210 is in 2.8.0 onwards btw ;)

gagarinlg, 2023-03-19:
Closed, fixed by #2210
We are not able to fix this via firmware update without users updating their firmware. 🤔

closed 2023-03-19 by gagarinlg

atagunov, 2023-08-18:
It looks like I'm having this with version 2.8.5 on BetaFPV LightRadio 3 Pro. Just cannot switch the radio on at all. Led lights up green but I never reach the stage at which there should be haptic feedback + sound and the screen remains black. Can still boot into bootloader mode and all looks normal then, can update from computer etc. There isn't an SD card in my radio is there? It seems to be using some sort of a chip instead. Should I open a new bug report?..