EdgeTX/edgetx #1567: Missing Antenna Parameter in Horus X10s

elekSENDER, 2022-02-06:
**Describe the bug**
There is no possibilty to change the antenna in the X10s.
Missing in X10s Radio Setting/Hardware and Model Setup.
In Companion this parameter is adjustable.

**To Reproduce**
screenshots from OTX 2.3.14

**Expected behavior**
adjustable antenna parameter in X10s

**Screenshots**
Companion EdgeTX:
[image]
[image]

EdgeTX 2.6.0
[image]
[image]

OTX 2.3.14
[image]
[image]

**General information**
 - EdgeTX version (2.6.0 x10-338d4d2)
 - Hardware (Frsky Horus X10s)

pfeerick, 2022-02-07:
Duplicate of https://github.com/EdgeTX/edgetx/issues/1185

elekSENDER, 2022-02-07:
ups, haven´t seen that, sorry for double-entry

PR #1765 (2022-10-11): X10/X12S: Int/Ext Antenna UI
Fixes #1567

Summary of changes:
- implements UI for internal/external antenna selection

And for prosperity/full transparency, I started this, but Raphael finished it ... you just can't see he finished re-writing the remainder of it due to the rebase. ;)

USE AT YOUR OWN RISK **until properly tested**! (aka have both antenna fitted until this warning removed).
closed 2022-10-11 by pfeerick