EdgeTX/edgetx #1574: "modelxx.yml" files not deleted.

Daannoa, 2022-02-07:
Hello,

**Describe the bug**
When I delete a model from my radio, the associated modelxx.yml file is not deleted from the MODEL folder. (The models.yml file is well updated.)

**To Reproduce**
Create or duplicate an existing model then delete it.

**Expected behavior**
delete the modelxx.yml file + update the models.yml file when deleting a model from the radio.

**General information**
 - EdgeTX version 2.6.0
 - Hardware : Radiomaster TX16S

pfeerick, 2022-02-08:
`moldelxx.bin` files were never deleted either, only the entry in the model index file... Why? I'm not sure exactly. It could be argued that it doesn't matter, if you don't fiddle with the files directly. It could also be argued that it means you can 'undelete' a model if you deleted it accidentally by simply re-adding it to the model index file. 

IMO, since you basically need a computer to do that anyway, Companion or a file copy should be the backup mechanism of choice, and thus the file *should* be deleted when you delete the model.

gagarinlg, 2023-03-18:
This is fixed since 2.8 for color LCD radios

closed 2023-03-18 by gagarinlg