EdgeTX/edgetx #1579: Tx16s internal bluetooth build issues

bsh314, 2022-02-09:
Hello! 

I was trying to build edgetx firmware myself using Gitpod the 2.6 branch with internal bluetooth module enabled using this parameters:
`cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DBLUETOOTH=YES -DCMAKE_BUILD_TYPE=Release ../`
after downloading the bin file and putting it on my tx16s sd card, i've flashed the bootloader and afterwards got into bootmenu to flash the firmware. After flashing the firmware my radio did not wanted to boot until I've flashed the latest prebuilt image from bootmenu again.

Do anyone sees any obvious error of mine here?

pfeerick, 2022-02-12:
Nothing obvious no. I suppose you want more than just bluetooth on, as I see HELI=NO flag is there also? Otherwise, see if the bluetooth build here works => https://files.portabledev.xyz/d/bf957dbe31f04e4a95a4/

bsh314, 2022-02-15:
I'll check if i can do that

bsh314, 2022-02-15:
> Nothing obvious no. I suppose you want more than just bluetooth on, as I see HELI=NO flag is there also? Otherwise, see if the bluetooth build here works => https://files.portabledev.xyz/d/bf957dbe31f04e4a95a4/

Thank you! just wanted to have Heli removed.

HazzaHFPV, 2022-02-20:
Is there an easy tutorial on how to compile EdgeTX? I'd like to compile it for my Jumper T16 with Bluetooth enabled but not quite sure how to go about it.

rotorman, 2022-02-20:
@HazzaHFPV See https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod and go with:
```
cmake -DPCB=X10 -DPCBREV=T16 -DDEFAULT_MODE=2 -DBLUETOOTH=YES -DCMAKE_BUILD_TYPE=Release ../
```

closed 2023-03-18 by gagarinlg