EdgeTX/edgetx #1584: [NV14] problem with external module update (ExpressLRS - R9M)

nightflasher, 2022-02-11:
i've switched over to edgeTX today and wanted to update the R9M module with latest and greatest ExpressLRS 2.2.0 - but there was this "no sync" error message and the module can't be updated.

yhomega, 2022-02-16:
I used a st link. Wasted too much time trying with opentx and nv14. Seems with edge tx it doesnt work too

nightflasher, 2022-02-16:
@yhomega this is not correct, it worked fine with the latest version of openTX by qba667 and stefan73.

stefan73, 2022-02-19:
Yes, it works fine on the OpenFlySky branch. I guess the reason is that the ext module driver is vastly incomplete.

Look at src/targets/nv14/extmodule_helper.cpp
And compare this to https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/blob/pulses23/radio/src/targets/nv14/extmodule_driver.cpp

This looks mergable to me.

raphaelcoeffic, 2022-03-23:
@stefan73 do you want to give it a shot? (PR?) You should know however that the external module driver has been completely re-written for horus & taranis target family, so that you might have to implement a completely different interface now. Also it should be converted to HAL/LL. Let me know if you need guidance.

raphaelcoeffic, 2023-02-14:
Guys, would you mind to retest this on the latest nightly? The complete module code has been rewritten and tested quite a fair bit. At least on EL18 this work quite good for me.

raphaelcoeffic, 2023-04-23:
@pfeerick should we close this one? (no feedback since Feb, and it seems to work for us)

gagarinlg, 2023-05-06:
Seems to work OK for us, so it is closed.
If someone still has this problem, please comment.

closed 2023-05-06 by gagarinlg