EdgeTX/edgetx #1585: External R9M ACCESS module dos not transmit channel information in f.port mode (telemetry works)

Zaquron, 2022-02-12:
I have a TX16s  with an R9M 2019 ACCESS module.
The receiver is connected to a Mateksys F411 Wing flight controller through f.port.
Coming from OTX 2.3.14, i installed EdgeTX 2.6 with externalaccessmod and RC output disappeared from the receiver.
When the receiver is connected, the channel data sometimes refreshes once and stays there until turned off. 
Telemetry always works fine.

Checked 3 times now, the whole setup works fine on OpenTX. 
tested with - flex mode (868MHz) with R9MX receiver
                   - EU (LBT) mode with the same receiver 
I guess there is something wrong with the externalaccessmod.

brainbubblersbest, 2022-02-19:
F.port is only a thing that happens between hf module and receiver. Hf module communication with Radio is the same as with sbus s.port configuration.
Are you shure that there is nothing wrong between receiver and flightcontroller?
Which FC Do you use and how have you connect the receiver with it.

Zaquron, 2022-02-19:
> F.port is only a thing that happens between hf module and receiver. Hf module communication with Radio is the same as with sbus s.port configuration. Are you shure that there is nothing wrong between receiver and flightcontroller? Which FC Do you use and how have you connect the receiver with it.

The FC is a Mateksys F411 wing. It's connected to the R9MX receiver through a serial TX pin.
I went back to OTX and everything works. There was no change in hardware or receiver side firmware in any way, that's why i think this is an EdgeTX bug.

brainbubblersbest, 2022-02-19:
I just like to request that you double checked This. 😉
Can you Check your Externalaccessmod wiring?
This is also a clear indication for Malfunctioning Inverter. 
(can be possible) 
TX to Module Bay Pin 1. 
Pin 2 is for Telemetry. 

You dont have a Scopemeter layin around somewhere? Or an old Accst Module and Receiver? 😅

Zaquron, 2022-02-19:
I'll recheck wiring tomorrow, but pretty sure it wouldn't work on OTX if it was a hardware issue.
I happen to have an oscilloscope, but no serial analizer. 
What should i check?
I could flash the module with ACCST but i don't have a compatible receiver so i can't test that.

brainbubblersbest, 2022-02-20:
You need no Serial analyzer.
Oscilloscope is enough. Check rx tx Pins in the Bay.
Signal should Look clean and sharp.
Maybe there isnt even a Signal on tx. (pin1) 

Every measurement thats done is good for bugtracking 😉

Zaquron, 2022-02-20:
I did a few measurements.
If by pin1 you mean the top pin, then there is a signal on that one. (running OTX or EdgeTX)
It does not look clean and sharp, but looks just like the signal i measured under OTX
[image]
.

scooper1, 2022-02-20:
That looks horrible. The levels are not consistant.   Try a higher sample if available on scope

Zaquron, 2022-02-20:
pin5 (the bottom one right under GND) looks like this:
[image]

Zaquron, 2022-02-20:
well, it's a very basic and cheap scope... 
Why does it work on OpenTX though?

brainbubblersbest, 2022-02-20:
The s.port Signal Looks good. The Signal on pin1 does not.
Idk why it works with opentx. Maybe @rotorman can say something about that. (different processing or else)
You can change module type to ppm to Check if you can Pick up a good Signal here.

Another idea would be to solder in an external inverter like on x10/x12. These are tested on 400k Baudrates.

rotorman, 2022-02-20:
@Zaquron can you show some pics of your hw-mod you did? The waveform in https://github.com/EdgeTX/edgetx/issues/1585#issuecomment-1046208365 looks pretty distorted to me.

Zaquron, 2022-02-20:
[image]
Connections look ok to me, but there is definitely a hardware issue here. 
When i connect the probe to pin1, the module loses telemetry even on OpenTX.
Now i have no idea how to fix this. :(  
Changing mainboard is beyond my budget.

Zaquron, 2022-02-20:
BTW just came back from the field where i had 2 flights on OpenTX. 
Not a single glitch,

rotorman, 2022-02-20:
On the USART3 header to the left of main STM32F4, I cannot clearly see from the image if the TX and RX wires connect or not - also flux should be washed off with PCB cleaner, as it is not only corrosive, but also lightly conductive.

Nevertheless, if with the same radio OpenTX firmware works OK, then there must be a software issue somewhere. I do not have R9M myself to test though.

Can you measure the top pin (PPM/SBUS of JR-bay) once more under OpenTX and under EdgeTX to see if you can spot a difference in the waveform?

Zaquron, 2022-02-20:
Resoldered and cleaned off flux:
[image]

Waveform on EdgeTX
[image]

Waveform on OpenTX
[image]

rotorman, 2022-02-20:
Solder job looks good, waveforms look the same. Hmm... would you have a logic analyzer, like Saleae, to check if the data stream from the radio is also similar?

Zaquron, 2022-02-20:
Sorry, i don't have a logic analyzer.

Zaquron, 2022-02-20:
Did anyone ever measure this with an oscilloscope? Maybe it's the same mess in every transmitter.

rotorman, 2022-02-20:
I had a look at TX16S schematics and the accessmod and I believe it would work better by removing 2 components highlighted below - continue only if you are adventurous enough:

[image]

[image]

Zaquron, 2022-02-20:
I have  a damaged mainboard (battery charge control circuit went up in smoke) so i might give this a try.
Does this affect anything else?

brainbubblersbest, 2022-02-20:
> Did anyone ever measure this with an oscilloscope? Maybe it's the same mess in every transmitter.

I only measured the waveform with th external Inverters on x12s. And ive done that with a fluke/philips pm97

rotorman, 2022-02-20:
> Does this affect anything else?
Removing the two components listed affects only the external module functionality.

Zaquron, 2022-02-21:
I did a little experiment. Changed the main PCB to the old one and did the suggested modification.
Signal still looks like this:

[image]

IMHO this is a software problem.
Can someone please take a look at the code?

ViorelX, 2022-02-23:
I have the same problem: T18 radio with R9M, which works fine in OpenTX, but it doesn't work in EdgeTX (only telemetry is transmitted.)

Zaquron, 2022-03-03:
Is there any chance someone with the skills will take a look at this? Please?

brainbubblersbest, 2022-03-03:
I would Do this. But I dont have access to a RM TX16S.

ViorelX, 2022-03-03:
The problem might be not specific to RM TX16s. I have Radioking T18.

brainbubblersbest, 2022-03-03:
Yes. First Post was rmtx16s.
I only have a X12s with additional Inverters.
And for me everything works.

raphaelcoeffic, 2022-03-22:
@Zaquron is that only with F.PORT output, or in general with this module? @brainbubblersbest can you please confirm: your R9M ACCESS works properly on the X12S with external access mod & edgetx?

brainbubblersbest, 2022-03-23:
I will Look into that this evening.

Zaquron, 2022-03-23:
> @Zaquron is that only with F.PORT output, or in general with this module?

I'll check as soon as i can find some time to experiment

brainbubblersbest, 2022-03-24:
[image]
[image]
The one with Radio is communication between Radio-module.
The other one is communication between FC-receiver, where yellow is sbus and blue is s.port.
I get vbat from fc and rxbatt from receiver.

In inav configurator i get only fixed channels. No Matter what i try to set in the cli.

I have old sbus s.port config over two separate uarts from the fc.

brainbubblersbest, 2022-03-24:
[image]
Input and outputs are defined in edgetx.
Values are changing in channel Monitor on Radio.

brainbubblersbest, 2022-03-24:
I try to figure out the last working build after come home from work. Dont know the last compiled version working with r9m in access.
I always need to self compile it.

raphaelcoeffic, 2022-03-24:
@brainbubblersbest ok, so same for you: connection seems alright (telemetry is coming, etc), but channels values are "fixed" and do not move.

brainbubblersbest, 2022-03-24:
Yes. And Signal looking pristine.
I can check with logic analyzer tho.

brainbubblersbest, 2022-03-24:
But... I just recognized the Signal doesnt Look pristine on the Radio photo. I dont realized it on live Monitoring.
Just right now on the photo.
The Signal Looks swapped after the half screen. I do another one with dc setting instead of ac.

brainbubblersbest, 2022-03-24:
@raphaelcoeffic Do you have a good scope Signal from access uart communication as it should be?
It came to my mind that this Digital Signal on TX Contakt is acting like ppm or cppm on that Single wave Form.

Can this be True? 🤔

ViorelX, 2022-03-24:
Today I checked if the channel data is transmitted in S.Bus & S.Port mode. No, it doesn't work in EdgeTX, only in OpenTX.

brainbubblersbest, 2022-03-24:
Found the protocols on opentx github.
Here is some short Video from the Signal, when moving the right Stick. (mode2) [Dropbox scope output](https://www.dropbox.com/s/teg5m1j39bofshv/vid-20220324-wa0014.mp4?dl=0)

brainbubblersbest, 2022-03-24:
@raphaelcoeffic  this seems related:
https://github.com/EdgeTX/edgetx/issues/260#issuecomment-987870988

brainbubblersbest, 2022-03-24:
aaand: can copy the exact behaviour as descripted.
F.port and sbus/s.port works.
but only under setting "fcc" 915/868/eu" wont work.
i´m using flex Firmware, so i should be able to choose what i want.

raphaelcoeffic, 2022-03-28:
@brainbubblersbest ok, so let me summarise:
- All 4 modes allow to bind successfully,
- but the channels only work with **FCC** (**915/868/EU** does not transmit channels properly).

brainbubblersbest, 2022-04-03:
Yes checked that again.
All True.
I can try to record 915/868/eu/fcc with logic analyzer.
If that is somehow helpfull.

pfeerick, 2022-04-03:
@brainbubblersbest I finally ordered a FrSky R9M 2019 /w R9MX receiver + a R9SX receiver so I can work on the FLEX UI stuff, so may also be able to help with this... (and/or will be asking you how to get started with it! :grin:)

raphaelcoeffic, 2022-04-04:
> @brainbubblersbest I finally ordered a FrSky R9M 2019 /w R9MX receiver + a R9SX receiver so I can work on the FLEX UI stuff, so may also be able to help with this... (and/or will be asking you how to get started with it! 😁) Will probably be a couple weeks before it arrives though.

You’ll need external access mod as well ;-)

Zaquron, 2022-05-02:
Any news on this? I'm itching to use EdgeTX but can't until it's fixed.

brainbubblersbest, 2022-05-14:
@pfeerick can you take a close Look into this please?
I know from a few users go back to opentx because of the non full working r9access function in edgetx. (inav /fixed wing /frsky group.

Momo-GitHub-1, 2022-05-16:
Working fine in my case with:
a) TX16s, R9M 2019 (ACCESS), R9MX, MAMBA APP F722, INAV 3.0.1.
b) as a), but with Matek stack in flying wing
edgetx 2.7.1 
cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DHARDWARE_EXTERNAL_ACCESS_MOD:BOOL=ON -DCMAKE_BUILD_TYPE=Release ../

Arvycka, 2022-05-16:
> Working fine in my case with: a) TX16s, R9M 2019 (ACCESS), R9MX, MAMBA APP F722, INAV 3.0.1. b) as a), but with Matek stack in flying wing edgetx 2.7.1 cmake -Wno-dev -DPCB=X10 -DPCBREV=TX16S [...]

in Flex mode 868MHz or 915 MHz?

Momo-GitHub-1, 2022-05-16:
> in Flex mode 868MHz or 915 MHz?
915MHz

federikk, 2022-05-16:
I had problem on flex 868, menu missing in tx ui, no servo movements

Zaquron, 2022-07-31:
who do i have to bribe to get this fixed? ;)

brainbubblersbest, 2022-07-31:
#2147 #2146
Hey @Zaquron I Think @pfeerick has ordered a R9m Module.
He's also my last Hope into this. I don't tried the functionality on the r9m as it seems that nothing changed in the Code expect the new GUi translation.

Arvycka, 2022-07-31:
Pre-2019 (non-ACCESS) R9M is working fine in EdgeTX. With ACCESS only EU f.port is messed up, if I recall corectly. I also went back to OpenTX because of this.

raphaelcoeffic, 2022-08-15:
As it seems, there was an error in the UI that could have caused what you see: normally, you should not have to choose the subtype (`FCC`/`EU`/`915`/`868`). Which frequencies / type is used is decided by the module FW (`FCC` vs `EU` vs `FLEX`) and in case of `FLEX`, which frequency you choose at binding time (pop-up menu).

So what happened is that the sub-type menu sets the `subType` attribute, which is then blindly sent to the module, which ignores the packets, are they are considered illegal (unknown sub-type bits).

**Summary**: the sub-type menu should be removed and `subType` must be set to `0` (which is what happens when `FCC` is selected in the sub-type menu).

Zaquron, 2022-08-23:
> As it seems, there was an error in the UI that could have caused what you see: normally, you should not have to choose the subtype (`FCC`/`EU`/`915`/`868`). Which frequencies / type is used is [...]

I can't wait to test this.
Hope it is merged soon.

Thanks a lot man!

raphaelcoeffic, 2022-08-24:
@Zaquron normally we test things *before* they get merged ;-)

Zaquron, 2022-08-24:
> @Zaquron normally we test things _before_ they get merged ;-)

Those are two separate sentences.  ;)

Though i'm still trying to figure out how i could compile a firmware for my transmitter with this fix included.
My knowledege is still very limited in this matter...

rotorman, 2022-08-24:
>i'm still trying to figure out how i could compile a firmware for my transmitter with this fix included. 

You can just fetch the GH built firmware under: https://github.com/EdgeTX/edgetx/pull/2211/checks

If you wish to build your own, you can use https://gitpod.io/#https://github.com/edgetx/edgetx/tree/r9m-access-fix-subtype w/o having to setup a local build environment (see for further instructions https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod ).

[7 more replies]

PR #2211 (2022-10-02): R9M access: fix subtype
Fixes #1585 

Summary of changes:
- do not send a sub-type for PXX2 R9M modules
closed 2022-10-02 by pfeerick