EdgeTX/edgetx #1587: Trainer Slave not recognized in Master.

dwhacks, 2022-02-13:
**Describe the bug**
x9 Lite does not detect when a slave trainer is plugged in (when configured as master)

**To Reproduce**
plug in a slave trainer with the x9 lite set to master

**Expected behavior**
Trainer connected status to be yes, and to accept a trainer

**General information**
 - EdgeTX version 2.5.0
 - Hardware Frsky x9 Lite

Here's the issue in OpenTX: https://github.com/opentx/opentx/issues/8670

pfeerick, 2022-02-13:
Is this fixed in OpenTX if you try a nightly? If so, it seems we made need following changes/fixes in our codebase also:
https://github.com/opentx/opentx/commit/2241c3f4ce3b0c6cd3fd41a45770c09ce3e75a7e (which appears to be what broke it in part)
https://github.com/opentx/opentx/pull/8811 (which appears to fix it for X7 Access and X9 Lite)

CaleyD, 2022-02-13:
May be completely unrelated, but I can’t get the zorro to recognize trainer port input either

pfeerick, 2022-02-13:
Most likely - I've not checked mine yet, but it probably suffers from the same trainer port socket issue as the TX12 and TX16S, where you either needed a specific extended plug, or to modify the socket to accept a normal plug.

dwhacks, 2022-02-24:
I haven't had a chance to test it (not a big fan of swapping firmware's back and forth) but apparently it has been fixed in opentx with https://github.com/opentx/opentx/pull/8811

P-I-Engineer, 2022-03-11:
was going to post an issue, but i'm having same problem. Can confirm that opentx 2.3.15 nightly the issue is resolved.

pfeerick, 2022-03-12:
Excellent. Thanks for that. I'll put a PR together that brings that across then. I'll call for testers again for the X9 Lite as I will be able to test it for the X9D+2019, but not the X9 Lite (or X7 Access).

dwhacks, 2022-03-12:
Perfect, I am ready to test it out on x9 lite!

PR #1685 (2022-03-18): fix: Trainer Slave not recognized in Master for ACCESS radios
Fixes #1587

Where Trainer Slave not recognized in Master due to trainer detect pin being inverted on ACCESS radios. 

Import of:
- https://github.com/opentx/opentx/pull/8811
- https://github.com/opentx/opentx/commit/2241c3f4ce3b0c6cd3fd41a45770c09ce3e75a7e

Currently untested. Needs testing on:
- [x] X9 Lite
- [x] X9D+2019 (previously not working RM WT01 to link slave transmitter worked simply by loading PR firmware)
- [ ] X7 ACCESS would be nice to have also
commit da482b48 (2022-03-18): fix: Trainer Slave not recognized in Master for ACCESS radios
Fixes #1587
closed 2022-03-18 by pfeerick