EdgeTX/edgetx #1592: Dismiss Keystuck Warning

jmxp69, 2022-02-13:
I just saw a comment on my discord: If the radio triggers a keystuck error, it cannot be bypassed meaning the user is locked out of the radio and cannot use hardware info to identify the offending key.

Two requests:
1) Warning page should identify which key is stuck.
2) User should be able to dismiss warning page so they aren't completely locked out of their radio without taking it apart and disconnecting switches to figure it out.

jmuxica, 2022-02-13:
I just experienced this on my TX16S (in OpenTX). 
I got the message "Key Stuck" with no way to know which key it is or a way around it. 
Further investigation lead me to the SA key.  I was able to get in, went through the HARDWARE menu ->  diagnostics -> keys and determined it was happening only with SA in the "up" position. When switching to the up position, a message comes up  saying that "Max Trim" had been reached. So every time I turn the radio back on, since the SA switch default is up, the message comes up at that point. Seems like something causes that trim to max out every time if the SA switch in on the up position.  I can't narrow it down any more.

pfeerick, 2022-02-14:
Except:
1. It _does_ tell you the key if it is stuck - a switch is NOT a key. 
2. It _can_ be dismissed - press enter, touch the arrow, or do nothing for five seconds and it will auto-dismiss (or instantly if the stuck key is unstuck)

What does seem to need to be improved on trim keys are not displayed if they are the stuck key/button, which in this cause was was likely due to the ribbon needing to be reseated. 

eg. 
https://youtu.be/NHhrq-PFUcI

[image]
[image]

jmuxica, 2022-02-14:
Thx for the info. Clearly the message on edgetx is different than the one I got in opentx.
The message in opentx says “key stuck” period. No more info and no more options.

jmuxica, 2022-02-14:
Btw this compels me further to migrate to edgetx sooner. 
But the suggestion remains: when delivering the warning message, additional information about which key is the one stuck should be included.

pfeerick, 2022-02-14:
Yup, but trims aside, this is a OpenTX problem ;) So don't close this issue as stuck trims not showing does need to be taken into account.

jmuxica, 2022-02-14:
Sure. .. Whether trim, key, slider, or switch I suppose.

pfeerick, 2022-02-14:
The "stuck key" warning is only for buttons - so trims and interface keys. Sliders and switches won't trigger it - that is the switches and pots warnings, which you have to specifically enable, and also tells you which switch/pot is triggering the warning. 

In your case it was rather deceiving hardware issue, as the misalignment of the ribbon cable was probably making SA hold the trim down...

jmuxica, 2022-02-14:
Got it! Thx for the explanation

gagarinlg, 2023-03-18:
key names are shown, trim names are not shown