EdgeTX/edgetx #1593: Slow Up/Slow Down and Delay Up/Delay Down Uses GVARS

jmxp69, 2022-02-13:
Using slow up/down delay up/down in the mixer requires hard coding a value in each field. Is there any reason those values can't use either a hard coded value as it does today or a GVAR?  

i.e. 
- If you want to slow down and slow up two different flaps, you have to adjust four fields on different pages. With GVARS this would be cut in half. Same with delay.
- If you want to slow down or sequence gear, changes would have to be made in two or three different mix pages. GVARS would greatly simplify this process.

elecpower, 2022-02-13:
Ideally GVARS should be available in all model fields.

B&W radio hardware limitations have tended to hinder implementation like why currently only 9 GVARS though some work to increase the number for colour radios has been toyed with https://github.com/EdgeTX/edgetx/pull/1189.

Yaml has removed the eeprom storage limitation for B&W though memory is still tight especially for those running lua scripts.

pfeerick, 2022-02-14:
> Ideally GVARS should be available in all model fields.

Indeed. If it doesn't add any major overhead to do so, I think we should work towards that... even if there are only 9GVARS for B&W... if you need all 9, well, you need to pick and choose what you use them for. Or if you only care about SLOW UP, SLOW DOWN, and a handful of other things, you have the option.

eshifri, 2022-02-14:
The fields that do not allow use of GVs are uint8_t or smaller.
GVs are saved at the end of the available data range (internally).
It means there are 2 options:
1. keep uint8_t and reduce the rabne of delays slow-downs to ~11 s.
2.  change the type to int16_t and increase the memory footprint (not a big deal, probably) and change storage and size-validation code. Not too complicated, but a lot of lines. :-)
The second solution will allow to increase the number of GVs as well if we increase the size of other fields (from int16::11 to full int16_t).

elecpower, 2022-02-14:
1. wouldn't like to give with one hand and take with the other.
2. It might not be much extra memory per field but when you multiply by the number of instances it will mount up. In the case of mixes there are some monster configurations out there with lots of lines.

Adding GVARS outside the field range can create conversion issues down the track if the range was to change. So one rule for all eg allow for storage of -1024 to 1024 and limit the settable range per field and GVARS -/+ outside; or make the field a int16_t union stealing one bit to indicate value or GVAR which should remove the conversion issue if need to expand storable range.

This is all good food for thought.

pfeerick, 2022-02-14:
It's not something we can rush into, ideally this would be any and all remaining fields that don't have GV as an option where it makes sense for them to have that as an option. If at the same time we can refactor code to make things simpler, that is a bonus.

The memory factor could be a real concern, as this issue is just focusing on the slow and delay fields... i.e. four fields PER mix line... so uint8_t to int16_t would mean 4x memory for those fields multiplied by the number of mixer lines, which _could_ render some currently working larger configs inoperable... 

Code-wise, this is indeed quite significant, as it means also means changes to the yaml config, companion, as well as quite a lot of testing will be needed. At the end of the day I think it is worth it... i.e. I think it is more a question of when, than if, this happens.

Tiefflieger68, 2024-08-17:
any progress on this?
in between things in code may changed to makes it easier to implement this nice feature...