EdgeTX/edgetx #1596: Module bay stopped working

DivingFalconFPV, 2022-02-15:
**Describe the bug**
Module stopped working with transmitter.  It worked for 3 weeks. Module(crossfire and elrs) powers on. And it does connect to drone receiver but transmitter doesn't pick it Up. Switches don't work. Agent lite and elrs lua scripts don't pick up module.  Agent m wifi shows module is connected to drone. 
Went back to opentx and module worked again. Went back to edgetx 2.6 and still didn't work.  

**To Reproduce**
Put edgetx 2.6 on. 

**Expected behavior**
Be able to use external bay. Control the drone.  Get telemetry. Lua scripts to pick up module

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version 2.6 
 - Hardware (e.g. Frsky X9D+ or RM TX12S)
 - Jumper T18 Pro v2
 - crossfire module and betafpv elrs24 module

brainbubblersbest, 2022-02-19:
Have you flashed a Binary with externalaccessmod?
Have you tried to lower your baudrate setting?

DivingFalconFPV, 2022-02-19:
> Have you flashed a Binary with externalaccessmod? Have you tried to lower your baudrate setting?

I don't know what binary with external access mod. I put 2.5 at first. It worked. I put 2.6 and it worked for 3 weeks. But some reason it just stopped. I haven't changed any settings. I can't figure out why. I refreshed firmware thinking it reset everything. But didn't work. 
Oh wait now u mention I changed buad to 5mil at some point. I can put edgetx back in see buad set

brainbubblersbest, 2022-02-19:
Baudrate setting affects s. Port communication.
Edgetx allowing way higher Baudrates as most Radios can deliver.
Try 400.000 or 115.200.
OpenTX cannot Do this. Thats might be your issue.
Please report back 😉

DivingFalconFPV, 2022-02-19:
> Baudrate setting affects s. Port communication. Edgetx allowing way higher Baudrates as most Radios can deliver. Try 400.000 or 115.200. OpenTX cannot Do this. Thats might be your issue. Please report back 😉

Omg thank you. I didn't even think of that till u mentioned it. Default it was working when I first set it up was 400k so I'll try that when I get home

gagarinlg, 2023-03-18:
Any news on this one?

closed 2023-05-05 by gagarinlg