EdgeTX/edgetx #1597: Auto Add import*.yml to models.yml

jmxp69, 2022-02-15:
The migration to yaml should open a lot of doors to portability. One of them is ease of sharing modelxx.yml files. Would it be possible to support an auto-import function of import-modelXXX.yml (or some other import triggering filename) so that if a user drops one or more import*.yml files into the MODELS directory the radio automatically renames the import-modelXXX.yml file to the next available model number, then adds the imported model into an easy-to-find location in models.yml--like the first slot?

It would be even better if all files with the name  import.modelXXX.yml were imported en masse so that all files with the convention import-model???.yml were automatically renamed and added to models.yml

I really don't care about the filename, the idea is to create a trigger so if the radio sees a import*.yml, it will trigger an import process to rename them and update models.yml for each import*.yml

raphaelcoeffic, 2022-03-23:
@jmxp69 I think this PR brings automatic scanning (the death of `models.yml`): https://github.com/EdgeTX/edgetx/pull/1374

jmxp69, 2022-03-23:
Does that scan automatically import new modelxx.yml files?

raphaelcoeffic, 2022-03-23:
> Does that scan automatically import new modelxx.yml files?

I think so. However you will have to take care yourself of renaming the files to avoid overwriting existing ones.

gagarinlg, 2023-03-18:
This is solved with the model selector introduced with 2.8

closed 2023-03-18 by gagarinlg