EdgeTX/edgetx #1605: [Flysky AFHDS 2A] MODE mismatch

nightflasher, 2022-02-18:
The modes in the flysky protocol (afhds 2a) are mixed up.
I have the nirvana NV14, where I first discovered this issue, so I switched from mode2 over to mode4. Today I've bound another receiver (fs-rx2a) to my jumper t-lite and discovered the same problem. Using  ExpressLRS for most of the time recently, there was no issues. 

My setting for both transmitters:
- RETA12
- MODE 2

FrSky on the t-lite works fine, ExpressLRS also works fine, FlySky modes are mixed up.

eshifri, 2022-02-18:
Can you please add some explanations?
Mode2 means certain mapping of the sticks to inputs. I assume this works for you correctly? (It does for me.)
RETA is a channel mapping. Is it broken? I use AETR without problems.
What exactly is broken? The default channels are not assigned according to RETA? If you change channel assignment in the mixers  does it fix the problem?

nightflasher, 2022-02-18:
The mode 2 is indeed the mapping of the sticks and it works in all other than FlySky Protokoll. The setting in the transmitter is default RETA12 and it's okay for frsky and ExpressLRS, but not for FlySky.
The channels are sent in wrong order.

eshifri, 2022-02-18:
What is the channel mapping (to sticks) you get?
No matter what configuration is the sticks should be mapped to first 4 channels.
As a work-around you can change the channel source in the mixer (RETA is just a default setting that can be changed for any model).
I'm using Mode2 and AETR - default FlySky settings and everything is correct.
I will try to play with your version later.

nightflasher, 2022-02-18:
Alter! Egal was ich verstelle, nur mode4 gibt mir das mode2 Verhalten!

Sorry! It doesn't matter what else I change, not reta, not AETR, not taer - but it only affects the flysky system!
Tried different receivers:
fs-x6b
fs-a8s
fs-rx2a
Crazybee F3 FlySky
All receivers the same!

And if I change the mode to 4 instead of mode 2, it works ONLY with the FlySky - the other protocols aren't affected by this issue.

nightflasher, 2022-02-18:
Here you can see, what is expected on the sticks with every node selected in the transmitter. If you're using FlySky protocol (AFHDS 2A), following happens:
set Mode 1 - get Mode 3
set Mode 3 - get Mode 1
set Mode 2 - get Mode 4
set Mode 4 - get Mode 2

This setting is set for the hole transmitter and every model and protocol, so changing to mode 4 to get mode 2 only in one model/protocol needs to be checked every time I want to fly any model. 

[image]

eshifri, 2022-02-18:
Are you creating a new model for every new set? Channel order setting affects only NEW models - for existing model it does not matter.
Receiver does not matter as well - you can see everything on the channel monitor.
Can you please check your mixers? What is channel assignment to the inputs?
I'm using NV14 in mode2 and it is working in mode2.

nightflasher, 2022-02-18:
If this only affects the NV14, I would be okay with this - but it's also on the jumper t-lite.
I only created one model and copy to get a new one with same settings. And changing the radio mode or channel order, affects all models - as expected!

nightflasher, 2022-02-19:
I've started from scratch, same behavior.

nightflasher, 2022-02-19:
FYI: all models on both transmitters based on the first one - the flysky one, but they all work the way it should...

eshifri, 2022-02-19:
Mystery...
Can you please take a look at the channel monitor on the TX itself?
One more crazy suggestion: change from PPM/sBUS to PWM/sBUS.

nightflasher, 2022-02-19:
You mean iBus?!

eshifri, 2022-02-19:
Doesn't matter. Whatever you are using with your controller.
Just use PWM instead of PPM.

eshifri, 2022-02-19:
Actually, I tried it: created  a new model with mode2 and RETA12 settings.
Channel monitor on the TX (NV14) is showing the correct reaction of the channels to the sticks.

nightflasher, 2022-02-20:
Channel monitor is okay, it's the data what is sent to the receiver and the flight controller.

eshifri, 2022-02-20:
Aha, now I can kind of reproduce it.\ even with PWM. Channels sent out are messed for everything, but AETR configuration.

nightflasher, 2022-02-22:
So you know where to dig into this?

eshifri, 2022-02-22:
Nope. :-)
Actually, I'm not sure it is a bug and not a feature.
Would be interesting to take a look at the same configuration in OTX.

nightflasher, 2022-02-23:
As I've never changed the default settings for the radio itself, it worked fine with every openTX release (by qba667 and Stefan73).

eshifri, 2022-02-23:
Same here. But ETX also works fine with the AETR setup.

nightflasher, 2022-02-23:
So you're recommended fix is to change this setting because it's the only working one!? So I need to change every quad and module? I doubt it!

eshifri, 2022-02-25:
If I read the code correctly (not sure) FlySky ignores the channel order in the system and always use the FlySky order - AETR.
This is for native AFHDS2A module - for Multi there is a mapping function.
Hopefully, somebody more knowledgeable in this are can take a look. @pfeerick ? @raphaelcoeffic ?   
Meanwhile the only thing you can do is to re-assign your mixies to AETR scheme.

pfeerick, 2022-02-25:
I would have to look at the code to know more re: NV14 specific behaviour or bug, but there are multiple issues present here due to the mix-mash of NV14 and MPM.  With the MPM, the channel remapping come into play

> RX output will match the Flysky standard AETR independently of the input configuration AETR, RETA... unless on OpenTX 2.3.3+ you use the "Disable channel mapping" feature on the GUI.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#FLYSKY-AFHDS2A---28

pfeerick, 2022-02-25:
Ok, after re-rereading this issue a couple more times, and also fiddled with a FlySky iA6B receiver, and it seems to be working as intended.

Lets stick with the NV14 for now, as the MPM + protocol dependent channel remapping option is a whole different issue (and just a matter of disabling it if you don't want it). Changing the mode setting is changing which stick maps to what source, and then that feeds into the order you have the channels defined in the mixer. So, if you just change from say mode 2 to 4 without changing anything else. With a TAER channel order, this means that the aileron would have first been on the right stick, but now it is on the left stick. I verified this is happening with a servo connected and changing the modes. Note that I'm also using TAER channel order, not the stock AETR - so that also indicates that the mixer channels are being output in the order I specify. 

So what is the actual issue on the NV14? As at the present time to appears to be working exactly as intended. If for a specific model you need the channels in a different order, your should be changing the order of the channels in the mixer for that model, not changing the radio wide stick mapping setting, because as you can see that buggers every other model up.

eshifri, 2022-02-25:
Hmmm....
I have created a new model with TAER and mode 2.
Receiver channel output was a complete mess.

eshifri, 2022-02-25:
Good/Interesting/Crazy discovery! :-)
I updated my SW to the current 2.7 code (from the version more or less corresponding to 2.6) and everything is working correctly now.

pfeerick, 2022-02-26:
Weird... I'm running a version of 2.7 from 1st Feb atm on the NV14, but there also hasn't been any real changes to the NV14 code AFAIK, or any of the channel order stuff, so more likely either something wrong in the config file, or needs more testing to determine what was going on... "Receiver channel output was a complete mess." isn't much detail to work from to try and figure out what was going on :-P

eshifri, 2022-02-26:
My code was built on Feb. 5-th.
It is difficult to describe the "mess": channel 1 was above 100%. Channels 2 and 4 were dead. Channel monitor was showing correct outputs.
TS provided some more details, but it looks like his problem was not exactly the same.

nightflasher, 2022-02-26:
Don't stick only with nirvana NV14, have the same issues with jumper t-lite (4in1 mpm).

nightflasher, 2022-02-28:
It doesn't make any sense. Why would everything work just fine except for the models with flysky afhds2 protocol, even this is the first model and any other model is based by "duplicate model" on this?

nightflasher, 2022-02-28:
WTF...you absolutely don't understand anything!!!
everything worked fine in openTX and it doesn't work in edgeTX - this applies to all the protocols and models! even those your scripts converted from openTX to edgeTX - except for flysky.
it doesn't matter which transmitter (ony have t-lite and nv14), both do the same wrong.

and of course i've changed the firmware of the mpm and it got worse! i've flashed back my reta mpm firmware and start changing the channel order in betaflight. this is kind of useless issue at all. 

i've found many things in your code for flysky specific which is totally different from the openTX branch (which works fine, even it is 2.2 or something like that).

pfeerick, 2022-02-28:
> WTF...you absolutely don't understand anything!!! 

This is not helping anyone. Everyone working on this project (including myself) are volunteering their time, so please be respectful even if you are frustrated. 

> everything worked fine in openTX and it doesn't work in edgeTX - this applies to all the protocols and models! even those your scripts converted from openTX to edgeTX - except for flysky. it doesn't matter which transmitter (ony have t-lite and nv14), both do the same wrong.

Firstly, yes, because there are some NV14 specific things we still need to port from the fork of opentx for the NV14, and it is only one of many transmitters being supported and worked on.

Hang on a minute, first you say "it's okay for frsky and ExpressLRS, but not for FlySky.", now you say it applies to all protocols and models... which is it? 

Why don't you rewind to the start, and actually say what it is exactly you are trying to achieve here? You have a betaflight flight controller, with channel mapping set to AETR1234. Flysky protocol, and you're using NV14(internal RF) and T-lite (MPM - RETA). Correct?

For the T-Lite, this means with MPM expecting RETA channel order input, and if you've left channel mapping turned on (MPM default), and channel order in the mixes page is RETA for that model (please verify, don't assume!!), the RF output order should be AETR. Correct?

So under exactly *what* circumstances does this not happen - is it when you change from model 2 to 4, without making any other changes. Precision is important, if you want this to be resolved. 

For the NV14, channel mapping does not apply, as that is a MPM only feature. So, the channels should be in the order you specify in the mixes pages of the model (NOT the default channel order, as that only applies to the creation of new models). 

Repeat testing on the NV14 (first with the 2.7.0 1st feb build I was running, then with 2.6.0 as second reference) with a FS-iA6B with RETA channel order , Mode 2, has CH1(R) moving when left stick moved side to side, CH2(E) moving when right stick up and down, CH3(T) moving when left stick up and down, and CH4(A) moving when right stick side to side.

Changing to Mode 4, CH1(R) is right stick side to side, CH2(E) is right stick up and down, CH3(T) is left stick up and down, and CH4(A) is left stick is side to side. This also looks exactly like you have your transmitter setup in https://github.com/EdgeTX/edgetx/issues/1605#issuecomment-1045087036, I get the same output on the outputs widget when moving the stick top right. And given the fact Betaflight shows the channels in AERT order (as opposed to RETA as shown on the transmitter, or AETR transmitted - note the T and R are swapped), that output on betaflight looks exactly right? Not mixed up, exactly as it should be as far as I can see.

Correct? If so, since I also checked Modes 1 and 3, it seems to me the NV14 is worked exactly as it should. So that means next need to look at the T-Lite + MPM, and see what is happening there.

nightflasher, 2022-03-01:
calming down - i've done exactly everything you've suggested. did it before several times: it's still the same behavior.
steps apply to nv14 AND t-lite:
- new firmware
- new model => flysky
- assigning my preferred switches to channels in mixer
- binding receiver (fs-x6b - NV14 | fs-rx2a - t-lite)

everything else is on it's default settings, as it always is and was. never messing around with channel mapping order in inputs or radio settings => MODE2 | RETA123 is the default

- now duplicate first model
- changing to external module (internal off)
- expresslrs binds up just fine (thx to binding phrase => love this feature)
- checking in betaflight => channel mapping works just fine

- duplicate once more
- changing to external RF to ghost
- module shows correct mapping

only on t-lite:
- duplicate first model (flysky) to model no 3
- changing to internal mpm => frsky
- binding to crazybee f3 fr tiny whoop (only one left with frsky)
- checking in betaflight => channels order just fine, as expected

- duplicate frksy to model no 4
- change RF both to off (simulator mode)
- attach cable
- fly simulator just fine

every other setting is default (Mode2 | RETA123)

what i've tried:
- changing modes in radio settings => Mode 4 is sent as Mode 2 | Mode 3 is sent as Mode 1 and vice versa
- -> affects all models, needs to be changed/checked whenever i change my model
- changing channel mapping order in radio settings => didn't change anything in flysky, but others (ELRS, frsky) are affected
- -> not a solution at all
- only on t-lite: flashed AETR firmware into MPM
- -> everything mixed up, couldn't get it to work with any of the created models
- changing channel mapping order in betaflight => from AETR123 to RETA123
- -> works fine, but needs to be paid attention to, whenever a new firmware comes out and if this will be flysky or ELRS

just to have it said: everything worked fine in openTX (both radios), so it must be something in flysky (afhds-2) protocol.
hope this clears everything up!?

nightflasher, 2022-03-13:
Find the same behavior on the SymaX protocol - don't know why.

pfeerick, 2022-03-14:
I haven't looked this again yet, but I noticed you said "flashed AETR firmware into MPM", when you are using RETA123 channel order. Because of the channel remapping that happens internally with the MPM for SymaX protocol (it requires the channels to be AETR order, which the MPM internal channel remapping takes care of), if you are still running the AETR input firmware, instead of the RETA firmware, the output channels will be completely scrambled. 

<html><body>
<h2 dir="auto">Symax - <em>10</em></h2>
<p dir="auto">Autobind protocol</p>

CH1 | CH2 | CH3 | CH4 | CH5 | CH6 | CH7 | CH8 | CH9
-- | -- | -- | -- | -- | -- | -- | -- | --
A | E | T | R | FLIP | RATES | PICTURE | VIDEO | HEADLESS

</body>
</html>

nightflasher, 2022-03-14:
@pfeerick nice try, but you've read wrong. I have TRIED another channel order, without any luck.
So everything still is RETA1234.

Sad to say so, but I'll go back to openTX on both transmitters - because i lately happened to have a bad accident with my dog and not working edgeTX as expected. 😢

closed 2023-05-05 by gagarinlg