EdgeTX/edgetx #1607: CRSF input for Trainer mode

sk8board, 2022-02-19:
Currently, the Trainer mode only allows PPM or SBUS input. I am requesting CRSF input for Trainer mode.

Here is why.

I place a receiver in the JR module bay and connect the receiver to the Batt, Ground, and HeartBeat terminals. If the receiver will output PPM or SBUS, then this will work for binding a student transmitter to the receiver in the JR module bay. Now wireless trainer mode can be used.

With internal ELRS modules becoming popular, now transmitters will have to use only ELRS receivers. The problem is ELRS receivers do not output PPM or SBUS. Therefore, there is a need for trainer mode to support CRSF protocol.

rotorman, 2022-03-23:
Luckily the channel data in CRSF is encoded 1-to-1 as with S.BUS. Here as a helper, a short CRSF parsing code by kobata, where he connected an HappyModel EP2 receiver via MCU to USB to act as a joystick: https://qiita.com/kobatan/items/40728fbb625057d9f42b

wimalopaan, 2022-03-23:
Does anyone have a full protocol spec at hand? For this the control-part (no sensor reply) would be sufficient. 

What would be the best test-setup for someone not owning CRSF hardware? Are the PC-based generators e.g. to use with some USB/serial converter?

rotorman, 2022-03-23:
Correct, you could just wire up the EP2 (CRSF) receiver TX line and leave the RX unconnected. Here a capture from the EP2 TX line (snippet):

```
C8 18 16 EC 03 9F 49 99 F3 0B F0 81 0F 7C E0 03 1F F8 28 08 00 00 44 1C D7 21 
C8 18 16 EC 03 9F 49 99 F3 0B F0 81 0F 7C E0 03 1F F8 28 08 00 00 44 1C D7 21
C8 18 16 EC 03 9F 49 99 F3 0B F0 81 0F 7C E0 03 1F F8 28 08 00 00 44 1C D7 21
C8 0C 14 35 00 64 0C 00 07 00 00 00 00 2E
C8 18 16 EC 03 9F 49 99 F3 0B F0 81 0F 7C E0 03 1F F8 28 08 00 00 44 1C D7 21 
C8 18 16 EC 03 9F 49 99 F3 0B F0 81 0F 7C E0 03 1F F8 28 08 00 00 44 1C D7 21
C8 18 16 EC 03 9F 49 99 F3 0B F0 81 0F 7C E0 03 1F F8 28 08 00 00 44 1C D7 21
```

wimalopaan, 2022-03-23:
> Correct, you could just wire up the EP2 (CRSF) receiver TX line and leave the RX unconnected. Here a capture from the EP2 TX line (snippet):
>
> ```
> C8 18 16 EC 03 9F 49 99 F3 0B F0 81 0F 7C E0 03 1F [...]

Cool, thanks for the data: will use that as replay ...

wimalopaan, 2022-03-23:
If I would like to get real HW, what would you suggest as RF-Module and RX? Available, simple and cheap...

wimalopaan, 2022-03-23:
The other protocols like SBUS, IBUS oder SumD/V3 use a fixed known baudrate. As I read CRSF can use different baudrates. But in the hardware-tab there is no more a baudrate selector.

What would be your suggestion how to implement that (at first):

1) use a fixed baudrate
2) use a user selectable baudrate (we need again a baudrate chooser)
3) autobaud (this may be only a try-n-error strategy by simple trying some well known baudrates.

rotorman, 2022-03-23:
Isn't the receiver side always 400k ?

wimalopaan, 2022-03-25:
In #1462  now we have

* sbus (tested)
* ibus (tested)
* crsf (not tested)
* sumd (not implemented)

wimalopaan, 2022-03-25:
For SUMDV3 I intend todo the following:

* first 16 channels are available as trainer input
* the binary switches 1-64 are mapped onto a LS iff the LS is of sticky type with no(!) set or reset condition.

So, if one wants to use the binary switches that come in via SUMDV3 trainer, she has to setup the wanted LS as above.

pfeerick, 2022-03-26:
I'll grab another ELRS RX out later to try the CRSF support...

wimalopaan, 2022-03-26:
> I'll grab another ELRS RX out later to try the CRSF support...

I do not guarantee for anything ;-)

pfeerick, 2022-03-28:
It seems to be working fine... only checked with the first four channels so far though :)

wimalopaan, 2022-03-28:
> It seems to be working fine... only checked with the first four channels so far though :)

Sounds good, thanks for testing.

sk8board, 2022-04-17:
Thank you for working on this. Do you know if this enhancement is targeted for a release?

pfeerick, 2022-04-17:
Should be in 2.8.

sk8board, 2022-10-12:
@pfeerick @raphaelcoeffic 

> Should be in 2.8.

Is CRSF input for Trainer mode still planned for 2.8? If so, can I test this with 2.8.0-rc3?

gagarinlg, 2022-10-12:
It is not going to be in 2.8.when you check PR #1462, you will see, that it needs to be rebased I clouding a bunch of manual work for translations.

pfeerick, 2022-10-12:
So #1861 supercedes and includes #1462?

wimalopaan, 2022-10-13:
Yes, it does.
It includes SBus/IBus/SumDV1-3 trainer input (up to 32 channels if EXTENDED_TRAINER is set) and FrSky-Telemetry-Input.

raphaelcoeffic, 2022-10-13:
> It includes SBus/IBus/SumDV1-3 trainer input (up to 32 channels if EXTENDED_TRAINER is set) and FrSky-Telemetry-Input.

Kind of a "pot pourri", we'll need to sort that out and extract features one by one.

wimalopaan, 2022-10-13:
Yes, it is. Actually I don't have time to sort that out by myself ... sorry.
But please don't hesitate to cherry pick the parts.

pfeerick, 2022-10-13:
Lol... have no fear, the skeleton will be picked clean, and credit given where credit is due ;)