EdgeTX/edgetx #1608: Preview log file on the screen to find lost model from last GPS...

zoran-grujic, 2022-02-19:
Hi,

If you go to SYS/SD-HC CARD/LOGS there is no option to read the log files, you can see they exist... In case you crash somewhere and you need GPS coordinates from the LOG file you are in trouble. 

Is it possible to view file contents like with TXT readme files? Well, CSV is actually just another TXT file? I know the LOG files could be bigggg, maybe there could be an option to read last 100 lines or so?

I tried to rename a file into TXT, but it is not possible to change its extension...

eshifri, 2022-02-19:
Can you please try renaming the file on the PC and try to open it in a text file viewer on the TX?
In OTX there is a limit of 2K  on the size of the text file, but in ETX it can be removed; at least on the color lcd radios.

brainbubblersbest, 2022-02-19:
Display should Show the last received coordinates.
Inav lua/ normal telemetry value.
You can make a Screenshot.
Btw. Can Screenshots be viewed in edgetx?

zoran-grujic, 2022-02-19:
> Can you please try renaming the file on the PC and try to open it in a text file viewer on the TX? In OTX there is a limit of 2K on the size of the text file, but in ETX it can be removed; at least on the color lcd radios.

I renamed one large file >1M, got View option and radio got stuck. I have TX16S. It was not possible to turn it off! I had to unplug the battery and it got back into emergency mode. After restart it was fine again. I was able to view 2k file, an 10k was OK but one need to scroll a lot.

Still the renaming strategy is not an option. You need another device...

zoran-grujic, 2022-02-19:
> Display should Show the last received coordinates. Inav lua/ normal telemetry value. 

Yes, but this might happen to be unreliable. 

>You can make a Screenshot. Btw. Can Screenshots be viewed in edgetx?

There is preview of a screenshot but it is too small to read small text from it.

eshifri, 2022-02-19:
Oops :-( Sorry to bring this problem on you.
This is a bug in its own right - the system should not try to open such big files.
If a shorter CSV file is renamed to TXT it is displayed normally.

zoran-grujic, 2022-02-19:
Yes, yes, and yes! 👍 

I am not going to submit the bug report, but someone probably should...

MartinHugh, 2022-02-20:
What is needed in this specific case (last known GPS) is for the function to simply read the last line of the log file, not the whole log file.

eshifri, 2022-02-20:
The thing that can be done easily is the following:
we can open CSV files of any size for review; we can open CSV files from the end.
Unfortunately, at the moment only on the color LCD TXs.

Do you think it will be useful/sufficient?

zoran-grujic, 2022-02-21:
In case last lines in log are corrupted I think opening CSV from the end is
the best option. But this must work for any size of the file!

In some more advanced future version I would suggest sending last
coordinates to users phone as path plus last heading and velocity. But this
is for future version...

pfeerick, 2022-02-23:
> In some more advanced future version I would suggest sending last coordinates to users phone as path plus last heading and velocity. But this is for future version...

Perhaps you would be interested in the following GPS widgets then? The second one may be of particular interest as it has it's own logger capability and viewer ;)
- https://github.com/kristjanbjarni/opentx-widgets
- https://github.com/moschotto/OpenTX_GPS_Telemetry

zoran-grujic, 2022-02-23:
I will give it a try.

But still, I would be happiest to read the end of the LOG file directly from EdgeTX. Text viewer already exists, it should not be too difficult to make a snippet of end of the log file either as temp file that would be open by text viewer or to develop an upgraded text viewer that will not crash everything on the RX when large file is open. 

I use INAV lua regularly to display telemetry info:
https://github.com/iNavFlight/OpenTX-Telemetry-Widget

PR #1625 (2022-02-26): TX viewer for log files for color LCD (#1608)
Fixes #1608 

Summary of changes: This PR adds menu option "View" for log (csv) files.
CSV files are opened in the TXT file viewer and opened from the end.
It allows to see last logged GPS coordinates.

This PR also includes fixies from #1613 , so #1613 can be skipped if this PR is accepted.