EdgeTX/edgetx #1611: Lua files compiled on EdgeTX 2.6 Companion incompatible with the Radio

pdxtigre, 2022-02-20:
I was trying to load the latest Yaapu 1.9.6 to X9D+ radio and ran into a couple issues.  One of that shows that the luac output from Companion when generated cannot be run on the radio.  

**"incompatible precompiled chunk" error**
[image]

The luac hex compiled on the radio vs companion shows similar issue mentioned in this thread
https://github.com/betaflight/betaflight-tx-lua-scripts/issues/281

Recompiled the lua source files again using OpenTX 2.3.14, then copied back to the radio with EdgeTX, I was able to run the app.

Additional info on why we cannot run Yaapu with EdgeTX on X9D+ can be found here
https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443/1653

gagarinlg, 2023-03-18:
The issue is, that companion and the simulator are now compiled in 64 bit mode, so Lua runs in 64 bit mode. This can only be resolved, when we switch to LUA 5.3, where the sizes can be forced.

gagarinlg, 2025-01-24:
fixed in #4203

closed 2025-01-24 by gagarinlg