EdgeTX/edgetx #1614: Edge TX2.6.0 Companion does not show/maintain custom telemetry screens

dk7xe, 2022-02-20:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
With ETX2.6.0 / Companion 2.6.0 custom screen maintenance does not work via Companion on Ubuntu 20.4 Linux.
Custom telemetry screen created on FrSky X9D+2019 is not showing up in Companion after loading models and setting. from Radio.
A Telemetry screen created within companion does not show up on the related model on FrSky X9D+2019 after writing Models to Radio.
It worked before with ETX2.5.0

**To Reproduce**
Flash ETX2.6.0 to Radio.
Create / Maintain a custom screen on a model.
Run Companion 2.6.0
Switch Radio on into boot loader mode and connect to PC.
Within Companion press read model an settings from radio button.
Open the model where custom screen was created/maintained.
Go to Custom screens in Companion and check content --> fields are empty.
Go to a different model that has no custom screen in Companion and set telemetry values to display.
Write model an settings to radio.
Check the related model custom screen --> no entries.

**Expected behavior**
custom screens maintained within ETX on radio should be visible within Companion and vice versa.

**Screenshots**
Telemetry screen of model
[image]
[image]
Screenshot companion
[image]

**General information**
 - EdgeTX version 2.6.0
 - Hardware FrSky X9D+2019
 
[image]

YAML File of the related model that shows no custom screen in Companion
[model05.zip](https://github.com/EdgeTX/edgetx/files/8103881/model05.zip)

men1al, 2022-02-20:
Exactly the same here, no custom screens in Companion 2.6:
- X9D+ with EdgeTX 2.6 (x9dp-338d4d2)
- Companion 2.6 Windows (Win11 Pro 21H2)

TX has custom telemetry screens which are working with EdgeTX 2.6, but are not shown by Companion 2.6.

It worked before with TX EdgeTX 2.5 and Companion 2.5 Windows version.

pfeerick, 2022-02-23:
Duplicate of https://github.com/EdgeTX/edgetx/issues/1557

dk7xe, 2022-02-25:
> Duplicate of #1557. Was fixed in #1562 which is and will be included in 2.6.1, which is delayed but will be out soon.

Peter, is there a nightly build of Companion available that can be tested? On https://github.com/EdgeTX/edgetx/releases/ i just see FW for the remote controls under nightly.

pfeerick, 2022-02-26:
No, only PR builds, as companion takes significantly longer to build. And unfortunately the last 2.6 companion branch build expired.

pfeerick, 2022-03-16:
Just a FYI, 2.6.1 is basically scrapped due to delays resulting in 2.7 release getting close, so you'll actually be able to grab companion builds from the main 2.7 branch now... since there has been some work on that... i.e. goto https://github.com/EdgeTX/edgetx/actions and on the left side pick the companion build action for your preferred OS, pick the latest of the builds from the `main` branch - i.e. there was one from three hours ago... and then get the download for under The "Artifacts" headings down the bottom of the page.

closed 2022-03-16 by pfeerick

dk7xe, 2022-03-23:
Thank you @pfeerick will test edgetx-cpn-linux-main https://github.com/EdgeTX/edgetx/commit/33f761c17b3b7b181c18e6ab2ab71f0670658a5e

dk7xe, 2022-04-04:
@pfeerick Peter i can confirm that with edgetx-cpn-linux-main https://github.com/EdgeTX/edgetx/commit/33f761c17b3b7b181c18e6ab2ab71f0670658a5e the problem has been solved. Custom screen maintenance works as expected.
@men1al fyi