EdgeTX/edgetx #1616: X20S Port

robthomson, 2022-02-20:
Ok.. can I point you to this page:

https://openrcforums.com/forum/viewtopic.php?f=7&t=13679

Seems MikeB has managed to work out 99% of the X20/X20S hardware base.

Can we consider a port?

brainbubblersbest, 2022-02-22:
Hello Rob.
As i wrote to you on FB:
MikeB has kind of a NDA with FRSKY.
Hes not allowed to Publish any Hardware Details FRSKY delivered to him.
I think @raphaelcoeffic would be open for that. But i think FRSKY doesnt want to share Technical Details with the other Radio Companys here.

robthomson, 2022-02-22:
I imagine this falls into a grey area.. Believe Mike may have looked at the assembler code to work out what inferfaces to use on the chip!

brainbubblersbest, 2022-02-22:
Maybe He is sort of a Ultra Brain, can read the Garbage output from a Decompiler and in addition He done the work to get this running in quite no Time.
There is still no Support from Frsky. Not for any Technical requests for Bug tracking. And no Details how to get the Tandem HF Hardware running.

I would love to see edgetx running on that Radio too. 😅👌

rotorman, 2022-02-22:
X20/X20S uses STM32H7 microcontroller. Presently EdgeTX uses StdPeriphLibs for low-level interface and it does not support STM32H7 (StdPeriphLibs support ends with STM32F4). Without a change in EdgeTX to STM32 HAL/LL Libs (planned towards v2.8: https://raw.githubusercontent.com/EdgeTX/edgetx.github.io/master/images/roadmap.png ), X20/X20S support is pretty much not possible yet.
Now please do not try to read between the lines of the above text - this does NOT mean that X20/X20S support is planned in v2.8 or afterwards - as w/o cooperation from the radio manufacturer, there is a considerable effort required to get the code running on such a radio. And with numerous other radio manufacturers with beautiful radios and genuine interest of having EdgeTX on their radios and support towards the project, the time of developers is better invested towards supporting these radios than wasting time with a manufacturer that has turned it's back to OTX/ETX.