EdgeTX/edgetx #1620: Backlight variable using Special Funtions on Horus 12S shows strange behavior.

GWguzzi, 2022-02-20:
**Describe the bug**
Move T5 down lowers brightness without problems to the lowest level.
Move T5 in the other direction increases the brightness to the maximum value, but with the last klick of T5, the brightness jumps to lowest level again.
The same behavior can be seen with T6. All other Inputs like slider L1 or L2 are working fine.

**To Reproduce**
Special function defined to dimm the display backlight.
Trimmer T5 changes brightness.

**Expected behavior**
no jump of brightness from highest to lowest level when T5 reaches the upper end.

**Screenshots**

**General information**
 - EdgeTX version: 2.6.0
 - Hardware: Frsky Horus 12S

pfeerick, 2022-02-23:
Reproducible on TX16s /w some random main/2.7 build...

Set BL SF value to any trim control (i.e. T5, ELE), and then use corresponding up to max out that trim channel, last step jumps from full to min brightness. Down is fine, only up has a problem. 

Tried on Zorro for comparision, and it suffers from the inverse... Up is fine, min brightness jumps from off to full on as last step... One way to wake up when testing at night! 😂

melvinchow, 2023-06-09:
This issue is still present on TX16S using EdgeTX v2.8.3